motioneye-project/motioneyeos #611: Build from source failure

virtualjp, 2016-11-23:
Hello,

I'm trying to build mootioneyeos from source but keep getting the below error during build.
I'm trying to build for raspberrypi and have tried raspberrypi2 and 3 as the board type and also bananapi just as an alternate board test but I still get the same failure,

I have tried building on a copy of Lubuntu(32bit) and Ubuntu Mate (32bit).
Could you shed some light on what may be causing this?
It's worth noting too that I am running the copy of Linux in an ESXi VM

---- error output ----
Makefile:3972: recipe for target 'all-gcc' failed
make[3]: *** [all-gcc] Error 2
Makefile:859: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:195: recipe for target '/home/jp/motioneyeos/output/bananapi/build/host-gcc-final-4.9.3/.stamp_built' failed
make[1]: *** [/home/jp/motioneyeos/output/bananapi/build/host-gcc-final-4.9.3/.stamp_built] Error 2
Makefile:36: recipe for target '_all' failed
make: *** [_all] Error 2
---- error output ----

ccrisan, 2016-11-24:
Please attach the entire log. I can't tell the problem from the small excerpt you have posted here.

virtualjp, 2016-11-28:
Hi Calin,

 

Thank you for your reply and apologies for delay response.

 

I managed to overcome the issue and believe it was something to do with my Buildroot configuration.

 

I am able to compile now from source, so all is sorted - Thanks

ccrisan, 2016-11-28:
Thanks for getting back. I'll close the issue then.

closed 2016-11-28 by ccrisan