motioneye-project/motioneyeos #612: saved wifi key does not persist through reboot

wanderingtechie, 2016-11-26:
Using RPI3, Version 20161114 . When I add a Wireless Network Key to the Network settings, following clicking "apply", the OS reboots and the key is not present.

ccrisan, 2016-11-26:
1. How does the Network section in the settings panel look like after a reboot? Please attach a screenshot.
2. How do you access your PI after losing the WiFi settings? Do you use a wired connection as well?
3. Can you attach your boot.log file?
4. Can you login remotely using SSH and check the contents of the file /data/etc/wpa_supplicant.conf?

wanderingtechie, 2016-11-26:
I am attaching the requested screenshot.

The system will become inaccessable if I boot it wifi-only. So I'll reboot
it with a wire.

[root@cam1 ~]# cat /etc/wpa_supplicant.conf
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
}
[root@cam1 ~]#

[root@cam1 ~]# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen
1000
    link/ether b8:27:eb:a6:9d:60 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.112/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast
qlen 1000
    link/ether b8:27:eb:f3:c8:35 brd ff:ff:ff:ff:ff:ff
[root@cam1 ~]#

Could this be because there are special characters in the wifi key?

> 1. How does the Network section in the settings panel look like after
>    a reboot? Please attach a screenshot.
>    2. How do you access your PI after losing the WiFi settings? Do you
>    use a wired [...]

-- 
----------------------------
Regards,
Michael Shulman
michael.shulman@gmail.com
Never attribute to malice that which can be adequately explained by
stupidity.

> 1. How does the Network section in the settings panel look like after
>    a reboot? Please attach a screenshot.
>    2. How do you access your PI after losing the WiFi settings? Do you
>    use a wired [...]

ccrisan, 2016-11-26:
It may be. What are the special characters you're using?

wanderingtechie, 2016-11-26:
The following symbols are in use:

{
[
^
_
+
*
(
)

riahc3, 2016-11-27:
I do not use special characters in my password.

I do have a number in my SSID.

ccrisan, 2016-12-04:
@wanderingtechie, your problem has been fixed with https://github.com/ccrisan/motioneye/commit/7dd8834bcebbca98361ffcbbb493b0317e2d4882. The fix will be included in the next motionEyeOS version.

closed 2016-12-04 by ccrisan