motioneye-project/motioneyeos #613: Camera ID field: Required

stllunatic, 2016-11-27:
I can't update settings or change the Camera ID field. It says "Make sure all the configuration options are valid!" when I apply changes. It says in a red circle that this field is required. Perhaps we could make it not required? The field is grayed out and I can't type anything.

I am using a Rpi3, usb-Philips_CE_Philips_SPC_1000NC_Webcam-video-index0, V4L2 Camera.
motionEye Version	20161125
Motion Version	4.0.1+git37b3595

Thanks!

snagglewest, 2016-11-27:
Same issue here

Rpi3
Logitech C210 USB Camera.
motionEye Version 	20161125
Motion Version 	4.0.1+git37b3595

ccrisan, 2016-11-27:
Have you tried clearing the browser cache?

stllunatic, 2016-11-27:
I've tried rebooting, removing the camera and then adding again, and from different devices I've logged in and I've had all of the same results. I have noticed that the problem is occurring in one build that has two cameras connected. It has the one listed above and this one.

usb-046d_HD_Pro_Webcam_C920_D7AC13FF-video-index0

In my single camera build it only has the camera listed below and it has assigned it camera ID 1 upon updating to the newest version of MotionEyeOS. (No issue)

usb-HD_Camera_Manufacturer_USB_2.0_Camera-video-index0

ccrisan, 2016-11-27:
You never answered my question. Have you tried clearing the browser cache?

stllunatic, 2016-11-27:
Clearing the browser cache worked. Thank you!

"Try a ctrl+f5 to clear the browser's cache as you might have been using an older version of motionEye before."

https://github.com/ccrisan/motioneyeos/issues/355

closed 2016-11-27 by ccrisan