motioneye-project/motioneyeos #614: Wi-Fi does not work

riahc3, 2016-11-27:
I add a SSID and a wireless password, system reboots and I cannot login to the system.

I have to reflash a new image in other to access MotionEyeOS again. 

Seems to me like Wi-Fi is broken.

Suggestion: Add a "Test connection" to the Wi-Fi section. Test it and if it works, THEN suggest a reboot so system reboots correctly and uses Wi-Fi instead of ethernet...or WITH ethernet...

riahc3, 2016-11-28:
Odd that I got a reply when I replied to something previously...but not now.

NicolasWoo, 2016-12-18:
I have the same issue with my Pi3 
Ethernet is working fine but it seems the Wifi is not working but I cannot figure out why...

those are the errors I am getting
 * Starting wpa_supplicant: failed (brcmfmac)
 * Configuring wireless network: no link
 * Configuring wired network: 192.168.0.40/24

Any ideas please

nmalley, 2016-12-19:
I'm having issues using a pizero and the Red bear bluetooth and wifi phat, configured the wifi over ethernet but no luck.  This was working on a previous version (not sure which as it was a few months back i last used it) flashed a new image last week - no wifi

NicolasWoo, 2016-12-20:
I managed to sort this problem by plugin a wifi dongle to my RasPi3...
Better than nothing

closed 2017-06-01 by bortek