motioneye-project/motioneyeos #615: Motion detection border and email

axum202, 2016-11-28:
Hello, 

Im trying to figure out why..
1.) There is no type of 'motion detection border' around the objects being detected....And
2.) Is this the reason why im not receiving email notifications with a attached snapshot of the object being detected? Even though the test emails DO work. 

Raspberry Pi 3, Raspbian Jessie. 

Great App by the way. Awesome .

axum202, 2016-11-28:
ps. i think its called the mask. Im not able to see a mask in my video. it is set to smart.

axum202, 2016-11-29:
I dont think its detecting motion. No red frame.

axum202, 2016-11-29:
I have hovered over the question marks and looked through the issues on this page. I hope its not a big problem. Any ideas? Thanks

ccrisan, 2016-11-29:
1. Does it detect any motion at all?
2. Do you receive notification emails at all?

axum202, 2016-11-29:
No.
No.

But when i sent a test email, it does work.

ccrisan, 2016-11-29:
What does your motion detection settings look like? A screenshot would help.

axum202, 2016-11-29:
Here are my motion detection settings.
Frame Change Threshold
0%-5%

Auto Noise Detection on
Light Switch Detection on
0%

Despeckle Filter on
Motion Gap 20 seconds
Captured Before 3 frames
Captured After 3 frames
Minimum Motion Frames 20 frames
Mask on
Mask Type smart
Smart Mask Sluggishness
5

Show Frame Changes off
Create Debug Media Files off

ccrisan, 2016-11-29:
What is your video device (capturing) framerate?

ccrisan, 2016-11-29:
Have you enabled working schedule?

axum202, 2016-11-29:
Yes , it is enabled to detect during working hours

ccrisan, 2016-11-29:
Try disabling the working schedule and see if motion is detected.

axum202, 2016-11-29:
framerate is at 2, I tried disabling work schedule and nothing

ccrisan, 2016-11-29:
With a framerate of 2 and a minimum motion frames set to 20, you'll need to generate at least 10 seconds of motion. Have you done that *after* disabling the working schedule?

axum202, 2016-11-29:
Yes, i just did that. Still nothing.

ccrisan, 2016-11-29:
Then please attach your configuration files (downloadable using the Backup button) *after* removing sensitive data from them. In need to see `motion.conf` and `thread-1.conf`.

axum202, 2016-11-29:
motion.conf 

# @enabled on
# @show_advanced on
# @normal_password 
# @admin_username admin
# @admin_password 
# @normal_username user

thread thread-1.conf
webcontrol_html_output on
webcontrol_port 7999
setup_mode off
webcontrol_localhost on

ccrisan, 2016-11-29:
Please attach them, don't paste them!

axum202, 2016-11-29:
[motion.txt](https://github.com/ccrisan/motioneyeos/files/619829/motion.txt)
[thread-1.txt](https://github.com/ccrisan/motioneyeos/files/619831/thread-1.txt)

ccrisan, 2016-11-29:
Your camera doesn't appear in your motion.conf. Please remove and re-add it.

axum202, 2016-11-29:
Ok I just removed and readded. Still nothing. After re-adding, should I change any settings? What should I see in motion.conf if my camera was added correctly?

ccrisan, 2016-11-29:
A line saying "thread thread-1.conf"

axum202, 2016-11-29:
I can see that line now. It is the last line of the file.

ccrisan, 2016-11-29:
Well, does the camera frame turn red when motion is detected now?

axum202, 2016-11-29:
No it doesnt. The camera is always focusing in and out, idk if that is a problem .

axum202, 2016-11-29:
Im also using this in my room , do i need to have much more space?

ccrisan, 2016-11-29:
set Show Frame Changes to on, to at least see that it does some detection and reduce the frame change threshold to something around 1%.

axum202, 2016-11-29:
Just changed those settings. Still nothing. By setting show frame rate to on, where would i find out that it is detecting motion?

ccrisan, 2016-11-29:
In the top right corner of the camera frame you should see a number. That's the number of motion pixels detected.

axum202, 2016-11-29:
all  see is frames per second in bottom left and timestamp in bottom right. camera name top left, and setting icons in top right.

axum202, 2016-11-29:
show frame changes is ON

axum202, 2016-11-29:
i have opencv installed on my pi. hope is has nothing to do with that.

ccrisan, 2016-11-29:
OpenCV shouldn't affect motion.

It's not on the overlay. You'll need to click on the camera frame once to make the overlay disappear. Then you'll see the clear camera image. On top of that you should see (in white) the number of frames changed in the top right corner.

ccrisan, 2016-11-29:
If that's not the case, please send me a screenshot of how your camera frame looks like.

axum202, 2016-11-29:
[image]

ccrisan, 2016-11-29:
This is odd. Please attach your two config files as they are now.

ccrisan, 2016-11-29:
Also what version of motion and motionEye do you use?

axum202, 2016-11-29:
[motion.txt](https://github.com/ccrisan/motioneyeos/files/620028/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/620029/motioneye.txt)
[thread-1.txt](https://github.com/ccrisan/motioneyeos/files/620030/thread-1.txt)

axum202, 2016-11-29:
motionEye Version	  0.35.1
Motion Version	4.0.1

ccrisan, 2016-11-29:
I'm sorry but things look a lot different here. It's like you weren't using motion 4.x. Can you please attach your motioneye log file (journactl -u motioneye > motioneye.log)?

axum202, 2016-11-29:
pi@raspberrypi:~ $ journalctl -u motioneye > motioneye.log
No journal files were found.

let me try to find the log file.

axum202, 2016-11-29:
nano motioneye.log file is empty

ccrisan, 2016-11-29:
How do you run motionEye?

axum202, 2016-11-29:
by going to my raspberry pi ip address:8765

axum202, 2016-11-29:
how should i run it?

ccrisan, 2016-11-29:
I meant how you *start* it, not how you open the user interface. Did you use the systemd unit and enabled it with systemctl? Did you use the init script?

axum202, 2016-11-29:
sudo systelctl start motioneye

ccrisan, 2016-11-29:
Then I cannot explain why you have no motioneye log. Could you give me temporary remote SSH access on your motionEye system, so I can see what's going on/

axum202, 2016-11-30:
yes, 192.168.116.130
pi@raspberrypi:~$

axum202, 2016-11-30:
there is no sensitive info in this device

axum202, 2016-11-30:
user : pi
password : raspberry

i will delete this info later

ccrisan, 2016-11-30:
That's your private IP address. You need to (1) set up port forwarding on your router and (2) give me your public IP address.

axum202, 2016-11-30:
I literally JUST figured it out....cant run motion and motioneyeos at the same time.

closed 2016-11-30 by axum202

ccrisan, 2016-11-30:
I'm glad you sorted it out :)