motioneye-project/motioneyeos #616: Camera not detected by motioneyeOS on RPi 2

mholland21, 2016-11-29:
So I flashed motioneyeOS to an SD card everything seems to be working (Wifi, etc) except the automatic camera detection for the on-board camera...

I know the camera works because I checked with raspistill before flashing motioneye to the SD card. I also tried using the Raspian + MotionEye method and could not get it to work (I'm a very novice coder) so I reflashed the OS. 

bcm2835_v412 shows up in lsmod but shows that it isn't being used by any other modules

[image]

I'm going to retry the raspbian method but I would still love some insight about my issue with the OS.

Thank you!

ccrisan, 2016-11-29:
What do you mean by "automatic camera detection". How do you know it doesn't work? Can you exemplify it with some screenshots?

mholland21, 2016-11-29:
In "local cameras" the camera doesn't show up. I'm currently double checking if the camera works in case something happened to the hardware. I'll post screenshots if I run into the same issue. Thanks for your quick response!

mholland21, 2016-11-29:
It seems as though my camera crapped out on me in my recent move as raspistill isnt detecting it anymore either (I last checked when i bought it a few months ago). Thanks again for your help, I guess I'll  have to buy a new camera and try again.

gitgoldberg, 2016-11-29:
I've several times had the tiny connector on the Raspberry Pi camera board pop open.  It can carefully be snapped closed again - but take anti-static precautions when doing so.
Sometimes just flexing the board when handling it is enough to un-pop the connector.

closed 2017-06-01 by bortek