motioneye-project/motioneyeos #617: "An error occured, refreshing is recommended" after trying to make long timelapse

pekkakorre, 2016-11-29:
Hi,
I tried to make timelapse from 900 pictures, 1 sec, 20fps and process go to full 100%, but download wont start and after that almost everything I do gets "An error occured, refreshing is recommended" message. Refresh and same error again and again. Reboot wont fix that, but update fixed it, until I tried again and same problem.

How can I fix that? Else than reinstalling?

I use RasPi 1 model B, motionEye version 20161125, motion version 4.0.1+git37b3595. 

Thanks from OS!

-Pekka

ccrisan, 2016-11-29:
Please attach your logs.

pekkakorre, 2016-11-30:
I dont think there is no need for other logs?

[motion.txt](https://github.com/ccrisan/motioneyeos/files/621077/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/621078/motioneye.txt)

ccrisan, 2016-11-30:
motioneye log file says:

    OSError: [Errno 12] Cannot allocate memory

Basically the video is too big to fit into memory. I recommend upgrading to a 1GB board.

closed 2016-11-30 by ccrisan

pekkakorre, 2016-11-30:
> Lähetetty: 30. marraskuuta 2016 11:19
> Vastaanottaja: ccrisan/motioneyeos
> Kopio: pekkakorre; Author

So can I get that error away or not? Do I have to reinstall and -configure whole system to get rid of that error?