motioneye-project/motioneyeos #618: Notifications

moonguar, 2016-11-29:
I am still getting notifications from the email service as well as the "Run a command" line even though I turned motion detection off on all my cameras.

ccrisan, 2016-11-29:
Can you please attach your configuration files after removing sensitive data from them? I need to see `thread-*.conf` and `motion.conf`.

moonguar, 2016-11-30:
Note that notifications (email and command line) are working correctly on the motioneyeos server which has one camera locally.  These files are from the motioneyeos server.  The problem seems to be coming from the motioneye client that has three camera's attached.  When I turn off motion on the motioneyeos server for the three cameras on the motioneye client, I still get email alerts and command line notices for all three of the cams on the motioneye client.  Interestingly, on another motioneye client that only has one camera attached, notifications are working correctly.  

motion.conf from the motioneyeos server
# @normal_password XXXXXXXX
# @enabled on
# @show_advanced on
# @admin_username admin
# @admin_password XXXXXXX
# @normal_username joseph

setup_mode off
webcontrol_localhost on
webcontrol_html_output on
webcontrol_port 7999

thread thread-5.conf

Here is thread-5.conf from the motioneyeos server

# @webcam_resolution 100
# @upload_subfolders on
# @upload_server
# @enabled on
# @network_server
# @upload_username
# @motion_detection off
# @upload_port
# @upload_location motion5
# @preserve_movies 7
# @network_username
# @upload_movie on
# @id 5
# @upload_password
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password
# @upload_service gdrive
# @name Camera5
# @preserve_pictures 7
# @storage_device custom-path
# @network_share_name
# @upload_enabled on
# @webcam_server_resize off
# @working_schedule

ffmpeg_output_movies off
height 200
stream_quality 85
threshold 2000
quality 85
noise_level 31
ffmpeg_output_debug_movies off
pre_capture 1
noise_tune on
smart_mask_speed 0
stream_maxrate 5
output_pictures off
hue 0
saturation 0
stream_localhost off
ffmpeg_variable_bitrate 75
ffmpeg_video_codec mp4
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
stream_port 8085
rotate 0
brightness 0
lightswitch 0
framerate 2
emulate_motion off
snapshot_filename
despeckle_filter
snapshot_interval 0
stream_auth_method 0
stream_motion off
target_dir /data/output/Camera5
text_double off
post_capture 1
stream_authentication joseph:XXXXXXXX
output_debug_pictures off
on_picture_save /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" picture_save %t %f
on_movie_end /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" movie_end %t %f
text_left Camera5
picture_filename
locate_motion_style redbox
locate_motion_mode off
contrast 0
videodevice /dev/video0
max_movie_time 0
on_event_end /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" start %t; /usr/bin/python /usr/lib/python2.7/site-packages/motioneye/meyectl.pyc sendmail -c /data/etc/motioneye.conf -l 'smtp.gmail.com' '587' 'XXXXXXX@gmail.com' 'XXXXXXX' 'True' 'XXXXXXX@gmail.com' 'XXXXXXX

width 320
@gmail.com' 'motion_start' '%t' '%Y-%m-%dT%H:%M:%S' '0'
event_gap 30
minimum_motion_frames 20
mask_file
width 320

ccrisan, 2016-11-30:
From what you have briefly described, I understand you have one motionEyeOS server and one motionEye client with 3 cameras attached. I assume the thread-5.conf here corresponds to one of the motionEye client cameras added on the motionEyeOS server.

If this is true, you have probably added it on motionEyeOS as a *Network Camera* (rather than as a *Remote motionEye Camera*). That's alright, but then motionEyeOS won't be able to remotely control the motionEye-specific stuff on your motionEye client; it will simply treat it like a... well, network camera.

Now if you have enabled notifications on your motionEye client, those must be disabled using the client's UI.

moonguar, 2016-11-30:
Calin,  thread-5 is the camera on my motioneyeOS server.  I did figure out the problem though.  I had the incorrect motion.conf file on my motioneye client.  I copied the motion.conf file from the client that was working correctly and modified it for the client it was copied to.  I restarted all processes and everything is happy and working correctly.  I did notice another problem though.  Should I start a new thread or should I just specify it here?  Thanks,  Kent

ccrisan, 2016-11-30:
Please open a new issue for every new issue :)

closed 2016-11-30 by ccrisan