motioneye-project/motioneyeos #620: Overclocking Issue

aldot99, 2016-12-01:
I have a problem.  By mistake I overclocked my Raspberry Pi 2 to TURBO.  Now the front-end interface is not responding and I have no way of accessing it.  The only thing that is still reachable is the Command line through SSH.  Are there any commands to overclok back to Pi2 setting through SSH.

ccrisan, 2016-12-01:
mount -o remount,rw /boot
    nano /boot/config.txt

Remove the overclocking lines and then press `Ctrl+O` (save), followed by an `Enter` (confirm) and then by a `Ctrl+X` (exit). Then you can `reboot`.

aldot99, 2016-12-01:
Thank you.  Thanks to your answer, the Raspberry Pi is now working fine on Rpi 2 Overclock.

ccrisan, 2016-12-01:
I'm glad I could help.

KarlBaumann, 2018-02-05:
I had the same problem. 
Maybe you should consider adding a warning, that Turbo mode does not work with RPI 2 or something similar

jasaw, 2018-02-05:
@KarlBaumann Overclocking is never guaranteed to work. It depends on the device itself. There will always be manufacturing tolerances, so sometimes you get a perfect chip that can overclock, sometimes you get a chip that is borderline working and not able to overclock at all.

KarlBaumann, 2018-02-05:
@jasaw agree! 
However, I was trying to make a bit different point in my comment - my RPI is in located in a hard accessible place - outside of my house mounted on the roof, and it was a bit annoying to climb up there just to delete these few lines from config.txt. 
If I would have known that there is a risk of "killing" my RPI, I wouldn't have played with the Overclocking settings. 
So I think it would make sense to add some kind a warning that this might happen and what to do in that situation.

jasaw, 2018-02-06:
@KarlBaumann I agree with showing a big fat warning when user clicks on the overclocking option. We might also be able to reduce the number of issues raised here that are related to overclocking.
@ccrisan What do you think?

ccrisan, 2018-02-06:
That makes sense. I'll consider adding a warning to the UI.

jasaw, 2018-02-06:
Alright, I've changed this issue to an enhancement.