motioneye-project/motioneyeos #623: Frame rate issues

Bobe001, 2016-12-03:
I am trying to use Motioneyeos to monitor a RTSP camera and stream the video out. The camera connects OK but I cannot get a steady frame rate - even on the preview - sometimes I only  get one or 2 frames per minute. Am I doing something silly or is Motioneye not able to do this?

ccrisan, 2016-12-03:
1. If you connect to the RTSP camera with a player like VLC, does the framerate seem better?
2. What have you set the Streaming Framerate (in the Video Streaming section) to, for this camera?
3. Have you toyed with preferences (in the settings menu, notably Framerate Dimmer)?

Bobe001, 2016-12-03:
Hi in answer to your questions
1. yes and the picture is rock solid, it also transcodes the stream properly.
2. Yes, and the video device frame rates to no avail, the fps monitor on the preview screen reads approx 7/4 at best dropping to 0.2/2
3. yes they make no difference.
It feels very much as though the preview and streaming outputs are being run through the motion detector, even though it is switched off.

ccrisan, 2016-12-03:
2. The answer "yes" is not valid for this question :)

What hardware is this? Have you tried to use a lower resolution? The streaming framerate might also be capped by the relatively low performance of the CPU.

Bobe001, 2016-12-04:
The hardware is Pi 2. the camera is an Onvid type. I have tried all different combinations of frame rate for both the Video device and the streamer all to no avail.. I suspect that the Pi 2 has enough computing power to deal with this task - the jerky screen trigger seems to be motion based..

closed 2017-06-01 by bortek