motioneye-project/motioneyeos #624: Audio output support on raspberry pi 3

bradbeam, 2016-12-04:
I started to poke around with enabling audio output support and it worked out to be pretty straightforward.

Here's the notes I wrote up from it. I'm not sure if there's better/more appropriate ways of doing this, but it worked for me to be able to play audio files via `aplay file.wav`

### Audio Support for the Raspberry Pi 3

To enable audio output support for the Raspberry Pi:

1. Remount `/` as rw  
    mount -o remount,rw /
1. Add the following to /etc/modules:  
    snd_bcm2835
1. Remount `/boot` as rw  
    mount -o remount,rw /boot
1. Add the following to /boot/config.txt  
    dtparam=audio=on
1. Reboot  

You can verify that everything is set up afterwards by running `aplay -l` and verifying that output devices are shown ( example output listed below ):

```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

moryoav, 2016-12-18:
If you can play then you can also record, and if you can record perhaps streaming the recording so you can hear what's being recorded from the motioneye control panel, maybe that's not too difficult to do as well? For years there's been talk in forums with people requesting audio support to be added to motioneye/motioneyeos, with the purpose of having a baby monitor of sorts. Having the audio streamed could be a very important first step...

buzink, 2017-02-05:
Can you not just do something like  `ffmpeg -f alsa -ac 1 -i hw:1 -ar 44100 -f mpegts udp://10.255.11.53:1234` ?

closed 2017-06-01 by bortek