motioneye-project/motioneyeos #625: Wi-Fi does not get IP Configuration from DHCP automatically

thnee, 2016-12-05:
I have configured the Wi-Fi correctly, and "IP Configuration" is set to "Automatic (DHCP)" in the web admin interface. This is confirmed by `iwconfig`:

```
wlan0     IEEE 802.11bgn  ESSID:"thnee2"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.457 GHz  Access Point: 00:24:B2:59:0C:B3
          Bit Rate:65 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=99/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.
```

After boot, `ip addr` shows:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether b8:27:eb:52:fd:73 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.7/24 brd 10.0.0.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether e4:6f:13:2b:cb:05 brd ff:ff:ff:ff:ff:ff
```

I then attempt to acquire DHCP config manually with `dhclient wlan0`.  
This works, `ip addr` now shows:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether b8:27:eb:52:fd:73 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.7/24 brd 10.0.0.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether e4:6f:13:2b:cb:05 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.116/24 brd 10.0.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
```

And I can access it successfully by the acquired IP `10.0.0.116` on the `wlan0` interface.

Another odd thing is that if I then proceed to disconnect the wired network, the wifi goes down again, becoming inaccessible. (I cannot show any output of this state, since I have no connection to the device). As soon as I plug back in the wired network cable, the wifi comes back up again and responds immediately.

ccrisan, 2016-12-05:
It appears that you want both wired and wireless connections to work at the same time. This is not currently possible with motionEyeOS. If you must use two or more network adapters at the same time, or have any other fancy network configuration, I recommend using Raspbian + motionEye.

thnee, 2016-12-05:
Oh it only supports one at a time? Well my goal was to only use wifi, not both. The wired connection was only used here for initial setup and trying to debug this issue. The core issue here is as stated: It does not get/use an IP Configuration via DHCP when booting, even without a wired network connected, rendering it inaccessible.

ccrisan, 2016-12-05:
Alright, then I'll reopen the issue. Please reboot *without* the ethernet cable connected, but with the WiFi properly configured. Then, I'd like to see the log files. It would be great if you could connect a monitor to your system so you can watch in real time whatever is going on there and if it says anything about connecting to the WiFi.

thnee, 2016-12-05:
Thank you! Yeah during boot it says `Starting wpa_supplicant: failed (8192cu)`, without any reason.

I tried enabling SSID Broadcast in the AP and that seems like it may have solved the problem. But I don't see why why that would be an issue, since the name and password are entered manually?

But if that is a known issue, maybe add it to https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

closed 2017-06-01 by bortek