motioneye-project/motioneyeos #626: SSL certificate problem: unable to get local issuer certificate

apbaseball, 2016-12-06:
I am unable to add any of my IP cameras using a JPEG or MJPEG stream over HTTPS. By default my cameras are setup with a self signed certificate over HTTPS (443). I am attempting to use the following for the stream:https://192.168.1.20/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=admin&pwd=password& and every time I get the following error: SSL certificate problem: unable to get local issuer certificate

Any thoughts as to how I can proceed with this? It is a Foscam FI9800P wireless 720p camera. I have also tried entering the RTSP stream information but nothing is recognized.

ccrisan, 2016-12-09:
@apbaseball have you tried setting [validate_certs](https://github.com/ccrisan/motioneye/wiki/Configuration-File#validate_certs) to `false` in `/data/etc/motioneye.conf`?

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

user9023523, 2019-04-08:
@ccrisan Setting validate_certs to false technically works but is there a proper way to do it? I tried putting the .pem in .etc/ssl/certs and rebooted but that didn't do anything. Am I putting something in the wrong folder?