motioneye-project/motioneyeos #627: Cannot access login page anymore - direct to full screen streaming

markvanhaze, 2016-12-06:
Strangely, after changing the port to 8081, using the browser I now do not get a login prompt anymore and instead a full screen stream of my webcam. How do I get back to the login page?

ccrisan, 2016-12-06:
Do not use port 8081 for the web frontend, as it's already used for the video streaming (which is what you see when you're accessing that port).

closed 2016-12-06 by ccrisan

markvanhaze, 2016-12-06:
Is there a way of changing the web frontend port without having to re-install the image?

ccrisan, 2016-12-06:
The web frontend doesn't start anymore due to the port conflict. You'll need to rewrite the OS image or login with SSH and manually edit the port in `/data/etc/motioneye.conf`.

markvanhaze, 2016-12-06:
SSH did it. Thank you for your help!