motioneye-project/motioneyeos #628: Monitor script losing execution bit

wandering213, 2016-12-06:
I'm not sure if this is a motioneye or motioneyeos issue, or possibly a hardware issue. 

I have a script to monitor whether or not my garage is open or closed, if works fine and displays properly on the garage camera. For some reason the file keeps losing it's execution bit, I've had to connect to the device 4-5 times already to modify the file to be executable. I had been running the script every 10 seconds without any issues, I wanted more frequent updates so I dropped that line in the script and then started having this issue. I've changed the delay back to 10 seconds and I'll see if that resolves the issue again.

Any other ideas?

ccrisan, 2016-12-06:
1. What version of motionEyeOS?
2. What board?
3. Where is the script located (full path)?
4. What line did you drop from the script?
5. What are now the contents of the script file?

wandering213, 2016-12-09:
1. 20161125
2. Raspberry Pi 3 
3. /data/etc/monitor_1
4. The final line of code: `echo 10 1>&2`
5. Here's the script contents, GPIO2 indicates whether or not a door is open. Then it checks/creates a file to track when the status changed last.:
`#!/bin/bash
status="$(cat /sys/class/gpio/gpio2/value)"
filename="/data/tmp/gopen.pid"
fileclose="/data/tmp/gclosed.pid"
if [ "$status" == "0" ]
then 
	if  [ -f $fileclose ]; then
                rm $fileclose
        fi
	if  [ ! -f $filename ]; then
		touch $filename
		chmod 0777 $filename
	fi	
	age="$((($(date +%s) - $(date +%s -r "$filename")) / 60))"
	echo "OPEN: $age MIN."
else
	if  [ -f $filename ]; then
                rm $filename
        fi	
	if  [ ! -f $fileclose ]; then
                touch $fileclose
                chmod 0777 $fileclose
        fi
	age="$((($(date +%s) - $(date +%s -r "$fileclose")) / 60))"
	echo "CLOSED: $age MIN."
fi
echo 10 1>&2
`

Since restoring the 10 second delay, I haven't run into the problem again.

ccrisan, 2016-12-09:
What if you say `echo 1 1>&2` instead of removing the last line completely? Does the issue still happen?

wandering213, 2016-12-11:
I made the change and I haven't run into the issue again since then.

ccrisan, 2016-12-13:
@wandering213 this seems like a voodoo thing to me. You should try narrowing down the problem, by first ruling out the script (that actually does some `chown` itself). Try minimizing the script contents (ideally removing chmods from it) and see if the problem persists. Other than this, I'm out of ideas.

wandering213, 2016-12-13:
@ccrisan That's basically how I feel about it at this point. I'll update the thread if I figure out anything new.

wandering213, 2016-12-17:
It's been a few days, removing the chmods appears to have resolved the problem

closed 2016-12-17 by wandering213