motioneye-project/motioneyeos #629: OS not booting on a Raspberry Pi 2 Model B v1.1

floehopper, 2016-12-06:
I can't seem to get the OS to boot on a Raspberry Pi 2 Model B v1.1 + Raspberry Pi Camera Module V2 + ThePiHut USB Wifi Adapter. I successfully wrote to the SD card using the following command on OSX: `sudo ./writeimage.sh -d /dev/disk2 -i ~/Downloads/motioneyeos-raspberrypi-20161027.img -n 'SSID:wifi-password`.

I obtained the `writeimage.sh` script by cloning the repo (i.e. it's the latest version in the `master` branch. Does that sound OK or do I need an older version compatible with the image? If so where should I get it from?

The same hardware boots fine with a fresh Raspbian (Jessie 2016-11-25) install. I can connect it to the wi-fi network and I have successfully used `raspistill` to capture an image from the camera, so I'm confident the hardware is all working OK.

What signs of life should I see when motionEyeOS boots up for the first time? e.g. should I see an open port 80 if I use `nmap`? Is anything output to the HDMI port? i.e. would I expect to see anything if I connect a display? Is there any way I can debug what's going wrong?

ccrisan, 2016-12-07:
If it's a raspberry pi 2, please try the raspberry pi 2 image.

floehopper, 2016-12-08:
@ccrisan: Doh! I didn't see the images for the different versions. I'll try what you suggest. Thank you.

closed 2016-12-08 by floehopper