motioneye-project/motioneyeos #630: rsync /data/ouptut of an motioneyeos device with a server

Rouzic6666, 2016-12-08:
Hi,
I would like to sync periodically  /data/ouptut of a motioneyeos rpi using a command as
rsync -avzh root@motioneyeos:/data/output local_folder
but rsync is not installed...
I tried to tinker an installation by copying files from a rpi where rsync worked without success...
How I can move forward
Regards
JL

cpinkham, 2016-12-08:
I have rsync running on a Pi running MotionEyeOS and I believe all I had to copy from a Raspbian Wheezy install was the rsync binary itself and the libacl.so.1 and libattr.so.1 libraries.  I put the binary in /usr/bin and the libraries in /lib and I can run rsync over ssh to pull files off the Pi.

Rouzic6666, 2016-12-09:
Hi cpinkham,
I had done exactly what you describe in your post... but may mistake was do use a "banana PI R1" rsync install as reference. It is not exactly the same processor 'A20 ARM Cortex-A7' versus 'ARMv6' for RPI  and generate error like 'illegal instruction' when executed... 
I redo the same from a RPI rsync install and, as you said, it works! thanks you!
JL

Chiny91, 2016-12-09:
Copying files across is hit and miss really - lucky it worked.   Linux generally gets upset if libraries/modules/etc get used that were not compiled against the same source.   The cleaner solution, if more complex, is to download the source for motionEyeOS from GitHub, config in rsync and re-compile.   I can say that works nicely, with no risk of oddball behaviour from mismatching libraries etc.

rsync is excellent of course for slurping those jpg files from the Pi.   I throttle rsync with --bwlimit to avoid too much stress on the bandwidth limited ethernet of the Raspberry Pi.

ccrisan, 2016-12-09:
Guys, 8a28d2c97adc3bcb7241b70751cd9000b77bcb0a adds the rsync binary to the OS image.

closed 2016-12-09 by ccrisan

Chiny91, 2016-12-10:
Excellent, many thanks; no more compiling specials for me :relieved:   My R & Banana Pi boxes never liked saving directly to Dropbox and so on; maybe a bandwidth throttle or raising the niceness of that process would have helped.   rsync is most useful for these aspects and behaves itself.