motioneye-project/motioneyeos #635: ./writeimage.sh under openSuse 13.2 doesn't work (for me)

CampoES, 2016-12-10:
Hi,

i've killed now two sd-cards by using the latest writeimage.sh - Script (motioneyeos-raspberrypi3-20161125.img). 

First I've to make the script executable by using "chmod +x motioneyeos-raspberrypi3-20161125.img". Than the code `sudo ./writeimage.sh -d /dev/sdb1 -i "motioneyeos-raspberrypi3-20161125.img"` start to write the image to the sd-card. 

The system give me the feedback that 220 + 0 blocks were written. After that something went wrong. I didn't save the messages so I try to explain it by my own words: The script tried to reread the image, but I believe, the sd-card is not mounted. Also umount of ./boot didn't is successful. I got the message that i do not use this system unless I made a reboot. After rebootinig the sd-card is not readable by linux :-( The command ls -l /dev/sd* didn't show me anything from the sd-card. lsusb the same. Nothing. 

Is there a chance to get a solution from someone ?

Thanks.

Michael

CampoES, 2016-12-10:
Hi again,

read in issue #179 that I do have to write the partition like this /dev/sdb and not like this /dev/sdb1. OK, works better. Now I got

****************
sudo ./writeimage.sh -d /dev/sdb -i "motioneyeos-raspberrypi3-20161125.img"
root's password:
:: writing disk image to sdcard
220+0 Datensätze ein
220+0 Datensätze aus
230686720 Bytes (231 MB) kopiert, 154,163 s, 1,5 MB/s
:: re-reading sdcard partition table
:: mounting sdcard
:: unmounting sdcard
rmdir: couldn't remove „./.boot“: Directory is not empty
****************

Now I wonder if the last error does matter?

(writing speed was slow because I took a cheap sd-card for testing)

Perhaps the above problem should be writen on the Instructions page?

Michael

CampoES, 2016-12-11:
And again me. 

Now I solved my problem. My cardreader was not able to read my 16 GB-card so I thought the card have an error. 
When I put the card in the SD-Slot of my Laptop I can access it. Then I format it again with ext4 an write the image to it. That works fine, but got the same message about "not possible to umount /boot". 

And the partition is only 180 MB!? Resized the partition to full cardsize an put it in my raspberry. Now it is working.

Michael

closed 2016-12-11 by CampoES

ccrisan, 2016-12-11:
@CampoES,

1. You don't need to format the card before. It's pointless as the `writeimage.sh` script will overwrite your partition layout anyway.
2. At first boot, the remaining space is automatically allocated to a third partition. You don't need to worry about those 180MB :)