motioneye-project/motioneyeos #636: Stream h264 in Fast Network Camera mode

tafryn, 2016-12-11:
The MJPEG stream supported in Fast Network Camera mode is very bandwidth intensive. Raspberry Pis can output a less bandwidth intensive h264 stream without a heavy CPU load using raspivid. Can streamEye be adapted to serve up this h264 stream?

Mikiya83, 2016-12-11:
In the same aspect, it should be perfect if h264 stream can be output even if not in "fast mode", when motion is on (as you said, MJPEG is very intensive for network)

ccrisan, 2016-12-11:
Adding a backend such as streamEye for h264 is definitely a good idea. However having motion stream h264 (i.e. "not fast mode") is not something I see happening anytime soon.

Mikiya83, 2016-12-12:
Ok, not a problem, it's already very good, thanks for your constant work 👍

mrgnou, 2017-11-12:
Adding h264 output to fast mode is a real need feature as it would allow support of high fps / definition video streaming which can't be done with mjpeg.

jasaw, 2017-11-12:
There are 2 different issues here:
1. Fast Network Camera mode stream H264, which can be implemented via uv4l and uv4l-raspicam.
2. Motion stream H264 (not in fast mode), which requires changes to motion software. This is difficult to implement and requires support from motion developers.

Mikiya83, 2017-11-13:
The first point could be very useful if it's possible to implement it quite easily !

jasaw, 2017-11-13:
@ccrisan Any thoughts on marking this issue as an enhancement request? The first point looks possible.

ccrisan, 2017-11-14:
I don't see any problem with it but I personally don't have time to take care of it. Pull requests are welcome, though :)

jasaw, 2017-11-14:
Alright, making this issue as an enhancement and anyone can submit a pull request if they want to.

d-faure, 2017-12-08:
According to https://www.raspberrypi.org/forums/viewtopic.php?t=51476#p449922,
or even https://chriscarey.com/blog/2017/04/30/achieving-high-frame-rate-with-a-raspberry-pi-camera-system/comment-page-1/
this should be doable combining `raspivid` and command-line `vlc`.

While the MotioneyeOS architecture is really nice as a camera firmware, this could be nice to embed `vlc` and provide a way to customize the streaming server to something else than `streameye` and/or `raspimjpeg.py`.

yonailo, 2018-09-27:
+1 to have this feature at least in fast-mode !!

Goeks1, 2018-10-07:
I am waiting also for a stream in h264 !!

jasaw, 2019-01-04:
See https://github.com/ccrisan/motioneyeos/pull/1764 and https://github.com/ccrisan/motioneyeos/issues/1259

jasaw, 2019-01-20:
If anyone is willing to be a guinea pig to test h264 stream in fast network cam mode, please go to PR #1764. There's a link to my pre-built image that supports this feature, but web UI hasn't been implemented yet, so it requires editing a couple of config files. Instructions are in the PR.
Any positive or negative feedback is welcomed.

Spacemarine2018, 2019-01-23:
I will try in the next days if possible. Finally one elegant way to watch my 3D-printer in high-res high FPS over low(er) bandwidth connection! Thanks!

jasaw, 2019-05-13:
Basic RTSP support has been implemented in dev branch.

closed 2019-05-13 by jasaw