motioneye-project/motioneyeos #637: Unable to open video device

SgtNobbyNobbs, 2016-12-11:
Hi,

I just installed the latest motioneyeOS image (20161211) on my brand new RPI3 but am unable to get video feed from the connected picamera v2; I see the camera in the viewer but the feed is gray and it says 'unable to open video device'. The camera works correctly, I tested that on vanilla Raspbian before installing motioneyeOS. 

I found a similar issue on this forum (https://github.com/ccrisan/motioneyeos/issues/147), but I think that was another problem, caused by connecting a non-compatible camera.

The 'motion' command causes a sort of loop, seemingly due to an issue with the camera: I see messages like 'device doesn't support VIDIOC_G_STD', 'error setting pixel format', 'VIDIOC_S_FMT: device or resource busy', and 'motion_loop: video device fatal error - closing video device'.

Could it be that PIcam v2 is not yet supported by motioneyeOS?

Any help is greatly appreciated.

Thanks!

ccrisan, 2016-12-11:
All RPi cameras should be supported by motionEyeOS. Could you please try this with a previous, stable release? It might be a RPi firmware issue.

SgtNobbyNobbs, 2016-12-11:
Thank you for your very quick answer!

Which stable version do you recommend?

ccrisan, 2016-12-11:
20161125

SgtNobbyNobbs, 2016-12-11:
I just tried the 20161125 build but got the same error message unfortunately.  Do you have another suggestion for me?

ccrisan, 2016-12-12:
At this point I'd recommend getting back to Raspbian and making sure it still works. The camera might have gotten damaged in the meantime.

SgtNobbyNobbs, 2016-12-16:
thanks for the answer! It turned out that my camera was indeed not working anymore. Issue solved!

closed 2016-12-16 by SgtNobbyNobbs