motioneye-project/motioneyeos #638: Login Credentials do not appear to apply

SmallConcern, 2016-12-11:
Hi

First off, thank you for all of your work on MotionEyeOS. It's fantastic. I'm setting it up on a Raspberry Pi 3, running the latest version (20161211)

I'm having an issue applying login credentials. I've set a password for admin and configured a surveillance login. However, upon applying and the pi rebooting, it will not accept my configured credentials with a password. It recognizes my change from 'user' to my new username for surveillance but accepts the login with no password. Same thing with admin, the password I set will not work, but empty still works.

I do see a message regarding Samba password on bootup, I'm unsure of its significance though:

> Setting smb admin password: Mismatch - password unchanged.

Thank you for any help

SmallConcern, 2016-12-11:
These debug messages are at the beginning of messages.log:

> Jan  1 00:00:27 meye-d89c43ec auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:27 meye-d89c43ec auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:27 meye-d89c43ec auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:27 meye-d89c43ec auth.err passwd: can't update password file /etc/passwd

ccrisan, 2016-12-12:
Things have changed a bit with regards to authentication in 20161211, indeed. The smb-related error is problematic. Those from messages.log, however, are normal.

1. Have you updated from a previous version? If yes, could you please try the new version from scratch?
2. Please clear your browser's cache, refresh and retry updating the password.

SmallConcern, 2016-12-12:
For each:
1. This was a fresh install today to 20161211. I have reformatted the SD card a few times trying earlier versions (20161125, 20161114). Interestingly enough, maybe related, maybe unrelated - after downgrading versions it would not let me login with the default credentials (admin/empty). It would not accept what I thought I changed the password to in 20161211 as well. When formatting up to the earliest version from the old, I did make sure to format the disc to empty before re-writing the latest image.
2. I tried clearing cache and trying a different browser on desktop and on my phone with the same results.

Would any logs be beneficial for debugging support?

koenig-dro, 2016-12-12:
Maybe Interesting to know:
After writing the image i could login as admin with no password after startup. Then nothing changed just reboot and it dosnt accept login as admin...

stefanoqredo, 2016-12-12:
same issue. very annoying. just a fresh install,  nothing else.
both admin and user works without password, I've tried to change (and rebooted) multiple times.

ccrisan, 2016-12-12:
This is indeed a nasty bug and needs to be addressed asap. I can reproduce it myself. Thanks for reporting it; I'll keep you posted.

ccrisan, 2016-12-12:
https://github.com/ccrisan/motioneyeos/releases/tag/20161212 fixes this.

closed 2016-12-12 by ccrisan

plauria, 2017-02-11:
Hi ccrisan,

Whatever is 'fixed' about this release (20161212) doesn't apply to the Raspi2 at least. I get the same behavior; changing passwd does nothing. Reboot and it's back to 'blank.'

Can someone else confirm?

plauria, 2017-02-11:
Sorry, my mistake. It appears that one cannot use 'passwd' with this OS but most go through the app. THis might be worth mentioning on the FAQ. Thanks