motioneye-project/motioneyeos #640: Access to picture signature in "Run A Command"

SmallConcern, 2016-12-13:
Calin,

Thank you for addressing the password issue so quickly. I'm running into a small snag that I was hoping you knew how to resolve. Is there a way to access information about the motion event that triggered a script? I'm working on integrating a script to push notifications to my phone and I'd really love to be able to provide a link to the image that triggered the event.

I can recreate the file path of the image that is created by using the time stamp options provided in command line arguments to Python - (%Y %m %d %H %M %S %q %v) - but I cannot get a handle to the signature created for the picture.

Is there an API available for what variables are available to the motion notifications actions?

PS: Is there a better format for this kind of discussion than creating Issues on the Github source? I'm a new to the Raspberry Pi scene and am not sure of the best place to ask these kinds of questions.

Thank you

ccrisan, 2016-12-13:
This is, I think, the best place to ask this kind of questions :)

Unfortunately there's no API to interact with motion(Eye(OS)). And unfortunately there's no easy way to obtain the pictures/movies associated with a particular event, other than the one you mentioned.

In fact that's the way the [sendmail.py](https://github.com/ccrisan/motioneye/blob/master/motioneye/sendmail.py) script currently does it and that's the way you should do it yourself (starting from sendmail.py as an example).

Alternatively you could try the `%f` special token when configuring the various "Run A Command" options but I'm afraid it may not work as expected.

SmallConcern, 2016-12-13:
I see! Okay I'll take a look at sendmail.py, I suspect that will get me what I need. However, I'd like to avoid having to move the file to another storage device to view it, is there a location on MotionEye(OS) I can place pictures such that they'll be viewable without credentials? Or perhaps leverage the browser cookie that MotionEyeOS already uses for the "Remember Me" functionality on log-in.

ccrisan, 2016-12-13:
You can do whatever you want in /data. There is no way those pics could be served via http, but you could use sftp, ftp or smb to access them. Sftp imposes limits, while ftp and smb will jail you to /data/media and /data/output. Still you can knock yourself out in those dirs :)

SmallConcern, 2016-12-13:
Okay, awesome. Follow up question: I was looking at sendemail.py and thought I might could leverage what you did with mediafiles to make my life a little easier, and noticed that motioneye is a pip install-able package (woot!), but I have tried getting pip to install but haven't had much success. I could download the source and run setup.py but it looks like it has external dependencies as well, and I have a feeling I would keep spiraling down the dependency tree trying to pull in source. 

I guess the root question is: is there a way to get pip installed on MotionEyeOS?

ccrisan, 2016-12-14:
1. You can't and you shouldn't install pip or anything related in motionEyeOS. If you want to install stuff, please use Raspbian + motionEye.
2. motionEye is pip-installable but it's already installed in motionEyeOS so I don't see why you would like to install it again.

closed 2017-06-01 by bortek