motioneye-project/motioneyeos #641: Dodgy SDcards

Chiny91, 2016-12-13:
Just taken delivery of 2 new Raspberry Pi 3s; both intended for motionEyeOS of course :grinning: These were bought as part of the official R Pi kit, complete with 2 off 16 GB SDcards.   I also bought 2 off 32 GB SDcards, ostensibly from a well-known manufacturer.   Everything was purchased via a very large, well-known supplier.

As usual I put all these SDcards through a soak test (on my iMac, software is F3).   The official kit 16 GB SDcards passed the tests flawlessly but the other 32 GB SDcards failed dismally.   The read/write speeds were low, the actual available memory was approx 14 GB (less than half) and even the supplied _micro to full size_ adaptors did not work.

Just a thought when you install motionEyeOS on that new SDcard, or put a new SDcard in your camera for that _once in a lifetime_ trip.

martinrgmiller, 2017-01-02:
The class of the SD card determines the speed. The RPi supplied cards are likely to be class 6 or better.
What class were the 32GB cards? I suspect you bought high capacity but slow cards.
https://www.sdcard.org/developers/overview/speed_class/

Chiny91, 2017-01-02:
The 32 GB SDcards were purchased as, and labelled as Class 10.

On arrival of the nominal identical replacements, I did notice the packaging was slightly different with features probably marginally more expensive to produce.   This could be later packaging, a different batch or real rather than "fake" SDcards.   I have no way of telling.

The replacement SDcards soak-tested perfectly and have run motionEyeOS flawlessly since installation.

closed 2017-06-01 by bortek