motioneye-project/motioneyeos #645: Can't see "Network Camera" when adding new camera

jexcobar, 2016-12-14:
Good afternoon, 

After update to release  20161125 and after to pre-release 20161212, when I'm adding a new camera, I can't see the Camera Type "Network camera", only I can see "Remote motionEye camera" and "Simple MJPEG camera"

Regards and tanks so much for your great work.

ccrisan, 2016-12-14:
Have you customized your motion binary? If yes, please revert it to the default (empty) or set it to /usr/bin/motion.

jexcobar, 2016-12-14:
No, I haven't costumized the motion binary. I only have updated to the releases I have written before.  I think,I'm going to write the SD again because I don't know what to do.

ccrisan, 2016-12-14:
Can you attach your log files here before wiping everything so we can at least see what the problem is?

jexcobar, 2016-12-14:
Here are:

************************motion.log********************
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-3.conf
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://192.168.1.52:82/mjpeg.cgi)
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://192.168.1.51:81/snapshot.cgi?user=admin&pwd=Admin1)
[1] [ERR] [NET] netcam_connect: connect returned error: Operation now in progress
[1] [ERR] [NET] Failed to open camera - check your config and that netcamera is online
[1] [ERR] [NET] netcam_http_request: Failed to read first camera header - giving up for now
[1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[2] [ERR] [NET] netcam_connect: connect returned error: Operation now in progress
[2] [ERR] [NET] Failed to open camera - check your config and that netcamera is online
[2] [ERR] [NET] netcam_http_request: Failed to read first camera header - giving up for now
[2] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)

jexcobar, 2016-12-14:
boot.log

---- booting ----
Setting hostname: done
Starting logging: done
Starting watchdog: done
Starting eudev: done
Setting up data partition
/dev/mmcblk0p3: Bad magic number in super-block while opening filesystem
Formatting data partition
Discarding device blocks:    4096/3859968               done                            
Creating filesystem with 3859968 4k blocks and 966656 inodes
Filesystem UUID: a7798666-4bfc-46dd-be07-10a02d28600e
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

Allocating group tables:   0/118       done                            
Writing inode tables:   0/118       done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information:   0/118       done

Setting empty admin password
Starting network: wlan0: no device
eth0: starting dhclient
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.137  netmask 255.255.255.0  broadcast 192.168.1.255
        ether b8:27:eb:af:0e:18  txqueuelen 1000  (Ethernet)
        RX packets 3  bytes 670 (670.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 700 (700.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 0  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

done
ssh-keygen: generating new host keys: RSA1 RSA DSA ECDSA ED25519 
Starting sshd: done
Setting current date using http: Mon Oct 12 07:55:28 UTC 2015
Starting http date updater: done
Starting proftpd: done
Starting crond: done
Setting smb admin password: done
Starting smbd: done
Starting nmbd: done
Starting motioneye: done
---- shutting down ----
Stopping logging: done
Stopping watchdog: done
Stopping eudev: done
Stopping network: done
Stopping sshd: done
Shutting down date updater: done
Stopping proftpd: done
Stopping crond: done
Shutting down smbd: done
Shutting down nmbd: done
Shutting down motioneye: done
---- booting ----
Setting hostname: done
Starting logging: done
Starting watchdog: done
Starting eudev: done
Setting up data partition
Setting custom root/admin password
[... 1589 lines cut ...]
 * Stopping syslogd: done
---- booting motionEyeOS 20161212----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.50/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Wed Dec 14 18:42:11 CET 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.50/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8

ccrisan, 2016-12-14:
Please attach them, don't paste them. This is madness!

jexcobar, 2016-12-14:
I'm sorry, here they are. Also I have attach a screen capture.

I have try add a new camera in several computers and browsers with the same result.

[image]
[motion.txt](https://github.com/ccrisan/motioneyeos/files/653389/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/653390/motioneye.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/653391/messages.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/653392/dmesg.txt)

ccrisan, 2016-12-16:
Well the logs indeed say that no motion binary could be found. Your SD card might have gotten corrupted. Have you solved this by reflashing the OS image from scratch?

jexcobar, 2016-12-17:
Yes, I have make a new install and all is now OK.
An I have tried the "Web Hook URL" - "Motion Notifications" with https://ifttt.com/ and it works very good.

Thanks again for your work.

ccrisan, 2016-12-17:
I'll close the issue then.

closed 2016-12-17 by ccrisan

TomPer, 2016-12-17:
Hello

I have exactly the same problem since 20161125 and after to pre-release 20161212.
Since my update, my 2 USB cameras are not detected. I will try to reinstall.

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/658864/motioneye.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/658865/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/658866/messages.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/658867/motion.txt)

ccrisan, 2016-12-17:
@TomPer don't reinstall. Just go to Expert Settings and clear the Motion Binary option.

TomPer, 2016-12-17:
Perfect !
Thank you !