motioneye-project/motioneyeos #646: Observations on use of MotionEye 20161125 on RPi B+ V1.2 , Cam 5MP V1.3, and Class of u-SD

JohnnyCantTube, 2016-12-15:
Calin

I struggled with the latest 20161125 build; at times my RPI B+ V1.2 would not boot
as I started to change expert settings.
I then discovered that CLASS 4 and CLASS 6 are not really fast enough for expert 
settings with MotionEye. Period.
I had so many 2GB left over from a number of Garmin NUVIs to re-purpose
but these older (micro) SD are just too slow.
I have tried so many different 5MP and 8MP(new) combos with 
RPI webcam OS and yours seems to be simple, yet feature rich, and cool.

I am attempting to encase my RPi surveillance 
(exterior mount, pointing down the driveway to watch for Courier Delivery)
to survive > -30 C lows, expected here(soon).
many thanks for your Brilliant work.

John R. in Ottawa Canada

martinrgmiller, 2017-01-02:
John
According to the Pi Foundation Class 6 should be good enough.
https://www.raspberrypi.org/documentation/installation/sd-cards.md

However, I've found that Branded Class 10 cards are still very cheap so I use the faster type. If you hunt around on Amazon you can get these for about C$5-10.

Regarding your external mounted camera, I've found the following to be highly beneficial.
1. use a sealed box. I use cheap IP61 rated boxes.  You shouldn't need better than IP65
2. leave the system running 24-7, the self heating of just a few Watts dissipated by the Pi and camera is sufficient to stop it freezing. I also have a 240V-5V converter inside my units which helps add a few more Watts.
3. add some pouches of silica gel in the spaces left in the box, this will absorb an moisture or condensation introduced in the build process.

Was your example picture taken from motioneye? How did you get the temperature on the picture?

Chiny91, 2017-01-02:
Great pic; especially as I have just had an email from a cousin in Ottawa grumbling about the cold :snowman:  I'd also like to know re that temperature text.

Here in the UK, I have never had freezing problems; admittedly it almost never drops below -5ºC in my location, once every few years, if that - I'm not sure it froze at all last winter.   My problem with warming the cameras (that infra-red for nighttime) is that spiders love the warmth, spin webs and a gentle breeze causes motion, with 1000s pictures.   The newish _smart mask_ has helped reduce that problem.

I concur with @martinrgmiller  re Class 10 cards; I always use them and it adds little if anything to the cost.

JohnnyCantTube, 2017-01-02:
@martinrgmiller : I have added some silica desiccant packs to the last external project. I still get frost on window of enclosure, but not as much since the addition of the IR LED ring (12V). My PoE setup using 12V (not 48) does not have enough power for two resistor "heaters" . so the experiments go onwards.
The picture is an SMC cam with the Temperature photo-superimposed for this screengrab, as part of the weather-rant we Canadians practice too much of.
The future objective to interface the motioneye cam "title" with the reading from a Maxim DS18B20 using GPIO4...  very good catch.

JohnnyCantTube, 2017-01-02:
@Chiny91 I had not yet thought of Spider but I too will need to seal up some apertures in those cases soon; thanks for the idea. 

in general, the a motioneye cam with weather resistant enclosure will head up to the family farm, 
where we hope the enhanced resolution will be enough to read license plates on the private entrance road. An SMC 640x480 cam mounted in the opposite direction proved inadequate to catch 'Holiday absence' thieves .

hud1nk0, 2017-02-07:
Hi, can i ask you, how did you manage to get temperature stamp with ds18b20?

closed 2017-06-01 by bortek