motioneye-project/motioneyeos #649: OrangePi zero

pit001, 2016-12-16:
It's possible to support the OrangePi zero (or Orange Pi One). A singleboardcomputer with ethernet, Wifi, USB for 6,99 USD (11 UDS incl. shipping). 

http://www.orangepi.org/orangepizero/

https://www.aliexpress.com/item/New-Orange-Pi-Zero-H2-Quad-Core-Open-source-development-board-beyond-Raspberry-Pi/32760774493.html?spm=2114.12010108.1000013.1.QShuX2&scm=1007.13339.33317.0&pvid=24f5796c-ef4a-40f8-9b8e-eeba11e453e0&tpp=1

peter

mictre65, 2016-12-16:
+1 on this request! Just picked up 2 OrangePi Zeros today actually. 
Thanks Calin for a very good software! 
/M

ccrisan, 2016-12-16:
The Orange PI Zero (and Orange PI PC) support is in the works. I hope to make it available with the next release.

mictre65, 2016-12-17:
Support of those devices would be much appreciated. Thanks again!

pit001, 2016-12-28:
great! thx!

xblox, 2017-01-01:
I am receiving my H2 next week too and give it a try. But yeah, I heard from a few angles that the Orange-PI company is nothing but a bunch of frivolous thieves, such guys shouldn't get any support outside of their rip-off ballpark. They have really bad reputation among the PI community.

ccrisan, 2017-01-01:
I have spent countless hours trying to port motionEyeOS to the Orange PI (kindly donated by a user). I have just given up and here are the reasons:

1. [Building](http://www.orangepi.org/Docs/Building.html) instructions point to missing repos on github. I couldn't figure out if they were renamed, moved, or just deleted.

2. [Setting up](http://www.orangepi.org/Docs/Building.html) instructions have no links or references to where one could download the required u-boot files. They pretend the files are somehow magically present in my home folder.

3. Both the legacy and mainline kernels have been wrapped into a single [repository](https://github.com/orangepi-xunlong/orangepi_h3_linux), making it impossible to instruct BuildRoot to procure kernel sources from the repo. I had to create my own [repo](https://github.com/ccrisan/linux-orangepi) for that purpose.

4. The mainline kernel has no camera module support. This is a deal breaker for motionEyeOS, forking me to go with the legacy 3.4 kernel.

5. The above all-in-one repo contains an enormous number of scripts in charge of building an entire OS for all the Orange PI boards in the world. I have looked through them and tried to apply what they do to motionEyeOS - I failed.

6. The "community" around these boards is a joke. Various efforts are scattered all around their crappy forum (that takes ages to load) and the [user-contributed instructions](http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=522) are incomplete, most of the time yielding errors.

7. There are many Orange PI boards, extremely poorly documented, making it impossible to identify possible compatibilites between them. If I somehow manage to build motionEyeOS for my Orange PI One board, I have no idea if it worked for any other Orange PI model.

8. After many many hours spent trying to make my OS boot on an Orange PI, I have finally managed to see u-boot telling its story, followed by the Linux kernel. What I wasn't able to do was to specify the root partition (/dev/mmcblk0p2 in motionEyeOS's case). The "official" [u-boot and boot scripts and fex files and whatnot](https://github.com/orangepi-xunlong/orangepi_h3_linux/tree/master/OrangePi-BuildLinux/orange) appear to ignore any uEnv.txt or boot.scr or whatever I place on the boot partition.

This being said, I am giving up on the Orange PI crappy boards. Unless anyone is willing to step up and  help me with the booting part, motionEyeOS will not run on these boards.

My work so far on Orange PI has been done on a [separate branch](https://github.com/ccrisan/motioneyeos/tree/orangepione).

xblox, 2017-01-01:
@ccrisan nice, thanks for the feedback. After your comments its now also crystal clear that the Orange-PI must go back in the oven. Its a plain insult indeed.

xblox, 2017-01-01:
Here a very good review: https://www.s-config.com/orange-pi-needs-to-go-back-into-the-oven/

mictre65, 2017-01-01:
@Crisan. Thanks for the efforts on MotionEyeos. I agree the documentation and download sites are a joke. However downloading the OpZero image Ubuntu image from the Armbian site and installing MotionEye (not the OS) on it works fine so Im quite happy.

pit001, 2017-01-01:
Yepp, I see. Useless hardware without a good documentation or a cummunity.
@Crisan. Thank you very much for your time, rearly !

ericjanvanputten, 2017-03-03:
@mictre65 Im trying to get motioneye on a orange pi zero now as well but get stuck around this part mkdir -p /etc/motioneye

tried multiple OSes.

xblox, 2017-03-04:
@ericjanvanputten : forget about the motioneye on orangepi just! Even if you get it running, it won't run stable. It crashes here randomly. I also installed a proper heatsink on it, changed camera, slimmed services,...

ericjanvanputten, 2017-03-04:
@xblox mmm good to know, I might still give it a go as I wanted to see if this would work better/cheaper then to buy 2x RPi3's

closed 2017-06-01 by bortek