motioneye-project/motioneyeos #650: Working Schedule

Rouzic6666, 2016-12-16:
Hi Calin,
I use the "Working Schedule" option. It works perfectly when range is included in a single day.
But when I try to define a period straddling two days at night to capture the event from 18:00 Monday to Tuesday at 09:00 it does nothing.
My configuration :
[image]
Thanks you for motioneyeos

Regards,
Jean Louis
motionEye Version 20161125
Motion Version 4.0.1+git37b3595

ccrisan, 2016-12-16:
It doesn't work that way. You cannot span working schedule across two days. The *from* field must always be smaller than the *to* field. Sorry about that.

closed 2016-12-16 by ccrisan

Rouzic6666, 2016-12-16:
thanks you,
I found the solution using complement...
[image]