motioneye-project/motioneyeos #651: Strange behavior of cameras

Bluscream, 2016-12-16:
I tested motioneyeOS for a few hours now and the two usb webcams i have connected start acting very strange. Sometimes one of them just says `UNABLE TO OPEN VIDEO DEVICE` all of the sudden for a few minutes before it shows up again. And the streams itself http://192.168.2.107:8081 and http://192.168.2.107:8082 won't work at all, but the streams in the main interface http://192.168.2.107 do.

Device: Raspberry Pi 3 Model B.
motionEye Version 	20161212
Motion Version 	4.0.1+git37b3595
Webcam1: /dev/v4l/by-id/usb-Creative_Technology_Ltd._Live__Cam_Sync_HD_VF0770-video-index0 (1280x720p)
Webcam2: /dev/v4l/by-id/usb-Novatek_Trust_Webcam-video-index0 (1920x1080p)

Full config file: [motioneye-config.tar.gz](https://github.com/ccrisan/motioneyeos/files/657134/motioneye-config.tar.gz)

LykkeGit, 2016-12-29:
I experience a similar issue. Did you solve the problem? 

I'm running a RPi 3 with an rpi OEM  Cam + a MPEG stream from an old Android phone (IP Webcam APP). After a while, the video streams disappear from the MotionEye UI and the 8081 port. Everything back to normal when I reboot the rpi.

Bluscream, 2016-12-29:
I only use one of the cameras. It's sad but with only one i can stream 25fps on 1280x720p without lags. Everything above causes meOS to act strange

LykkeGit, 2016-12-29:
Exactly! I have had the pi running with one cam for weeks and thought it was time to upgrade. Will post if I find the silver bullet.

LykkeGit, 2016-12-30:
MotionEye has now been running overnight w. two cameras without issues.

My problem seems related to me saving the images to a ReadyNas share, i.e. when I configure it for local storage or upload to Google Drive everything runs smoothly. Odd..

Bluscream, 2016-12-30:
Hmm very strange im not saving anything amd still have the issues

closed 2017-06-01 by bortek