motioneye-project/motioneyeos #652: Can't Post to Google Drive

KiwiNibbler, 2016-12-17:
I have three MotionEye cameras.  All are configured virtually identically.  One of them stopped uploading to Google Drive a few months ago.  It still takes photos.  I have tried it using wifi and cabled.  The only difference in the three is that the one that doesn't upload has a different camera.  Aside from that - identical.  The one with problems also seems to use more disk space on the SD card.  It is using 4.9Gb.  The other two are each using 1.3Gb.  It was up to 12Gb, but I deleted old photos.  That seems like a lot of photos but it takes about the same number as the others.  At the same quality.  So disk usage should be about the same.  Yes?

Testing the service fails.  It just shows the spinning circles for a minute or so then quits with no result shown.

ccrisan, 2016-12-18:
Please attach your motionEye log file from the camera that does not work.

KiwiNibbler, 2016-12-19:
I can't.  When I click on the log files, it just locks up.  Is it possible
to telnet to it?

Thank you

ccrisan, 2016-12-19:
Use SFTP and look for `/var/log/motioneye.log`.

hansen-m, 2017-01-03:
I just wanted to chime in with a "me too". I haven't had a chance to debug this but my Google Drive uploads seemed to stop working around Dec 14, so I suspect either the most recent update or a change in the APIs caused something to break. I only running 1 Raspberry Pi B+ model.

I tried obtaining new API keys and updating the upload paths without any luck. Clicking the "Test Service" button confirms the keys are working. The images and videos are being saved to the SD card but not uploaded.

KiwiNibbler, 2017-01-10:
Hi,

Just getting back to troubleshooting.  Still no luck on retrieving log
files.  It locks up when I try to download one.  I took a config from one
of my other cameras and applied them to the faulty one.  It didn't help.  I
can't get it to respond to SFTP (Or any other protocol for that matter).
Any suggestions?

Also, it is using a huge amount of storage compared to my other cameras.
My others use about 500 - 600 Mb.  This one is sitting at 11.2 Gb.  It
isn't photos or video using the space.  Not sure what it is since I can't
get into the OS.

Cheers!!

KiwiNibbler, 2017-01-10:
I gave up and wiped the SD card and started from scratch. Worse than ever.

Several issues:
- The latest release on GitHub is 20150719.  I tried it and it failed:
- I can setup the settings with no issues.  However, I know that there are new versions.  If I do a check, it says that it is up to date.  I know that is wrong.  I have enabled pre-releases but that doesn't help.
- If I change the IP address to the IP address that it had previously, all settings go away and I cannot make changes.  Also, the admin password is reset to something I cannot guess.  I had to reformat the SD card and start again - a few times.
- The option to save to Google or DropBox is not there.  I suspect it is because I have an old version.
- Now it just constantly reboots.  Can't log into it, can't stop the reboots.

I found an old copy of 20160109 from a harddrive in a drawer.  I will wipe and re-install using that version.

KiwiNibbler, 2017-01-10:
Completely back online now.  Used the build I had (20160109) then updated to latest pre-release.  One odd thing still.  If I use the IP address it had originally, it loses its mind.  Admin password changes, settings go away, etc.  Mildly annoying.  Will need to change my other cameras IP addresses to match.

hansen-m, 2017-01-19:
Looking at the motioneye.log file this appears to be the only line that may be relevant -

`2017-01-19 16:25:52: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=2&_signature=7e3a6fc8ddf82d8a97cc7365f93514debc0fb6e7 (127.0.0.1) 100.18ms`

hansen-m, 2017-02-13:
No noticeable changes after updating to version 20170212

dandodex, 2017-02-15:
Interesting, I lost my ability to upload to Google Drive and Dropbox around December. I wonder what has changed.

2017-02-15 14:40:55: [motioneye]    ERROR: could not read upload services state from file "/data/etc/uploadservices.json": No JSON object could be decoded

My uploadservices.json file exists with date Dec 30 17:52, but it is empty.

dandodex, 2017-02-15:
Nothing like having to explain what my problem is to be able to fix it myself.

I deleted the empty /data/etc/uploadservices.json file, restarted, re-authorised getting a new Dropbox key, entered it in the GUI and it is not working fine and uploading everything!

closed 2017-06-01 by bortek