motioneye-project/motioneyeos #655: All Pictures twice...

Dobi29, 2016-12-18:
Hi,
i have all pictures twice in my folder.
These are my settings:
[image]
[image]
[image]
What's wrong ?
Thx
Klaus

ccrisan, 2016-12-18:
How can you have the pictures twice? Duplicate filenames are not allowed.

Dobi29, 2016-12-18:
i have them twice uploaded on my google drive
[image]

ccrisan, 2016-12-18:
Please add a -%q at the end of the still images filename pattern and see if the problem persists.

Dobi29, 2016-12-19:
-->
[image]

Chiny91, 2016-12-19:
It looks as though you have the video device (camera) set at Frame Rate 2 (per second).   Google and motion are correctly storing both pictures, although with a slightly different file naming mechanism.   If you want only 1 pic/sec, you'll need to drop the frame rate.

ccrisan, 2016-12-19:
That is correct, your motion is configured to store more than one picture per second. While that is not something I'd recommend, it is possible, but you'll have to make sure the filename pattern allows it (which is what you did by adding %q to it).

closed 2016-12-19 by ccrisan

Dobi29, 2016-12-21:
...but it is the same picture.

Chiny91, 2016-12-21:
How do you know it is the same (what is your "Frame Change Threshold") ?   Try enabling "Show Frame Changes" in "Motion Detection" and then look at 2 pictures of the same second.