motioneye-project/motioneyeos #657: Video playback

aldot99, 2016-12-19:
I am currently using a Raspberry pi 2 with a CSI camera board.  The Pi is overclocked at the setting "Pi2" and the GPU memory assigned is 150.  The videos recorded are in .avi format and the movie quality at the moment is set to 100%.
My issue is the playback of the recorded movies that are saved onto a Network drive.  The Playback speed appears to be too fast and in my opinion this is caused by the "bottleneck" effect.  However, the resolution, the framerate and other options are set to the minimum.  Could it be the camera?
I am not sure whether my CSI camera is the authentic verison since I have bought it on ebay.  
[image]

ccrisan, 2016-12-19:
I'd recommend setting the movie quality to about 50%. The file size (and required bandwidth) will be greatly reduced.

aldot99, 2016-12-19:
Thank you for your response.  However, could it be that the "fake" CSI camera is slowing down the process and not picking up the frames?

ccrisan, 2016-12-19:
I have no idea. I have experienced myself fast movies and I believe it has to do with not enough resources and frame skipping. But this is a [motion](https://github.com/Motion-Project/motion) thing and I wouldn't know how to help you with it other than advising to reduce the movie quality and save the file locally so there's no network issue.

closed 2017-06-01 by bortek