motioneye-project/motioneyeos #659: Motion is gone after the upgrade to 20161212

rafi1975, 2016-12-20:
Hi,

I just performed the upgrade to 20161212 from 	20160919 and lost the camera output completely.
I can see the following in the logs:

2016-12-20 21:51:55: [motioneye]    ERROR: motion not running, starting it
2016-12-20 21:51:55: [motioneye]    ERROR: failed to start motion: motion executable could not be found
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 323, in checker
  File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 98, in start
Exception: motion executable could not be found
2016-12-20 21:51:56: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:51:57: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:51:58: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:51:59: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:52:01: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:52:02: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:52:03: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:52:04: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2016-12-20 21:52:05: [motioneye]    ERROR: motion not running, starting it
2016-12-20 21:52:05: [motioneye]    ERROR: failed to start motion: motion executable could not be found

Which is a bit of a pain as I have to claim to the PI very high on the wall to get it out and install from scratch.

Thanks,

Rafal

ccrisan, 2016-12-21:
Make sure your Expert Settings/Motion Binary option is empty or set to `/usr/bin/motion`

rafi1975, 2016-12-21:
That did the trick. On all of my Pi's the path was /usr/bin/motion/mr-dave.

Thanks

closed 2016-12-22 by ccrisan