motioneye-project/motioneyeos #660: Email Notification Setting Up

MaxfieldXie, 2016-12-21:
Hello everyone.  When trying to set up email notification for motion detection. it kept failing. Please help! Thank you in advance!
[image]

I have avoided special characters in my password, it is a new gmail account 
[image]
This is the message after hitting the "Test Email"  I have enable Still Images 
[image]

ccrisan, 2016-12-21:
Are you sure gmail's SMTP allows you to use a different account than your email address?

ccrisan, 2016-12-21:
Also, I believe that Google's SMTP uses port 587 by default with TLS.

MaxfieldXie, 2016-12-21:
Sorry, Can you help me understand a lit more when you said " gmail's SMTP allows you to use a different account than your email address?" True be told told, I don't know how to set it up. 

after I changed the port number new error message 
[image]
Than you for the quick response !

closed 2016-12-21 by MaxfieldXie

BillSmith68, 2019-01-20:
That feature doesn't work with Gmail. There is always some sort of error.