motioneye-project/motioneyeos #664: Volume Name has space in it

BarryIA, 2016-12-26:
Hi Folks!

May have found an issue with installation, or at least I was unable to find the problem already listed.  If the Volume Name has a space character in it will cause installation to fail at Line 76.

Using a Ubuntu 16.04 machine to install the image file from a Thumbdrive to the SD Card (via USB adapter).  Reason for the Thumbdrive is I’m new enough to Linux to not be certain of what  I’m doing; wanted some ‘extra checks’ to keep me from formatting my hard drive accidentally.  

Used two thumbdrives, one with the pre-installed volume of “ADATA UFD” and later another with “USB Thumbdrive”.  Both had a installation failure with the error listing Line 76.  The error also added a ‘1’ to the device (/dev/sdd became listed as dev/sdd1), which caused a ‘not found’ error.  Renamed to “USBThumbdrive” (eliminated the space character) and installation went smoothly.

I’m not sure ‘whose’ Line 76 is.  Used the ‘writeimage.sh’ script to assist with the installation.  The error occurred with both MotionEyeOS-RaspberryPi2_20161125.img.gz and MotionEyeOS-RaspberryPi_20161212.img.gz.  (I thought for sure I had a RPi2 Model B; really was the Model B – oops!) 

Apologies in advance for the vagueness and missing details..  I was doing this in mid-December and finally getting around to reporting the issue about two weeks later.

ccrisan, 2016-12-27:
I don't think the volume label has anything to do with motionEyeOS at all. Please post a log of the writeimage.sh script if it fails.

BarryIA, 2016-12-28:
Hi Calin!  

Experimented and now I'm confused!  Worked fine with the space in the volume name -- seemed like that was the problem with the first try.  Did do a couple of things differently, partially since I was experimenting anyway: as got my WiPi today added the wireless configuration (-n option) and a static IP (-s option).  

So, not sure what happened but seems to be working fine now.

Thanks for a great programme and have a Happy New Year!

Barry

ccrisan, 2016-12-28:
Alright, I'll close the issue then.

closed 2016-12-28 by ccrisan