motioneye-project/motioneyeos #668: Using Chrome trying to change camera name fails with "An error occurred. Refreshing is recommended"

silversword411, 2016-12-28:
Testing with Chrome, and logged in with admin ID gets error (even after reboot) when trying to change the camera name. 

Renaming with Edge worked fine. Haven't tested other browsers.

ccrisan, 2016-12-28:
Clear your chrome cache and refresh.

silversword411, 2016-12-29:
Tested from Incognito window, same result (which is basically a blank cache). Chrome version 55.0.2883.87 m

ccrisan, 2016-12-29:
1. What version of motionEyeOS?
2. What have you tried changing the camera name into? Any special characters? Spaces?
3. What error do you get? I love screenshots.
4. If it's a server-side undexpected error, please attach your motioneye log file.

silversword411, 2016-12-29:
MotionEye Version 20161125
Just appending number or letter (anything, no special characters I've tried just single letters as well)
[image]

Attached is a snip of the end of the motioneye.log file, don't see anything useful there.
[motioneye1.txt](https://github.com/ccrisan/motioneyeos/files/677136/motioneye1.txt)

Offtopic: Is there a way to clear the log files? Just delete them?

ccrisan, 2016-12-29:
This doesn't make any sense. The "an error occurred..." message is always accompanied by an error in the log file. Are you completely sure you have downloaded the log *after* generating this error in the screenshot? It says `2016-12-29 11:10:54`; is this your local time about now?

silversword411, 2016-12-29:
Here's one with debug on, then off, then on again. 
Definitely log file downloaded after trying to apply name change (did it twice). Adjusted framerate dimmer first to write to log. Then changed name and clicked apply. Then downloaded log file.

[motioneye3.txt](https://github.com/ccrisan/motioneyeos/files/677167/motioneye3.txt)

Something in the javascript of the page maybe?

ccrisan, 2016-12-29:
This log is even more confusing. Timestamps are around "2016-12-29 08:36:59". So again, what's the local time now there? It also says that you have configured a static IP of "192.168.1.101" while I can see it in your browser that you accessed it using 10.0.8.201. Do you have some kind of routing/nat between the two networks? Are you definitely sure you're downloading the logs from the machine that generates this error?

silversword411, 2016-12-29:
The local time previously wasn't matching with the device (which was on auto update from google during the first 1-2 messages in this thread). Had to change to NTP, then back to http from google before it would update. motioneye3.txt logs reflect the correct time of ~8:34am EST and forward.

Not sure why that's reading static IP in 192.168.1.x subnet. It's on wifi DHCP

[image]

There's no 192.168.1.x network, it's all 10.0.8.x. This devices has had several upgrade versions loaded overtop.

Couldn't login with SSH when in debug mode (not sure why, log was saying ssh=True). Finally just pulled the MicroSD and reformatted fresh with motionEye version 20161212. Got back into ssh.

So far, all I've done is: Set timezone, enabled expert mode, set wifi name and pass ,turned ftp off, smb auth off. ssh on. Raspberry Pi3, connected now via wifi and ethernet. Here's the log again after trying to set the name:
[motioneyeclean1.txt](https://github.com/ccrisan/motioneyeos/files/677777/motioneyeclean1.txt)

[image]

silversword411, 2016-12-29:
FYI network config: 
[image]

ccrisan, 2016-12-29:
Could you please open the developer tools in chrome *before* changing the name and follow the output of the dev tools console while the error occurs? This is to make sure the error is a consequence of a bad AJAX or something like that.

silversword411, 2016-12-29:
Not sure if I'm looking in the right place but does this answer the question?

https://youtu.be/h1L56TeLaPk

ccrisan, 2016-12-30:
This is exactly what I asked for. And it's very weird. I was expecting an error in the console associated with the message.

Would you be able to set up remote access for me so I can try to debug the issue?

silversword411, 2016-12-30:
Ok, just testing from another PC and no problem. Apparently Chrome incognito isn't as blank a slate as one thinks. :/ Thanks for all the followup on this, sorry for the wasted time :(

closed 2016-12-30 by silversword411

john72oliver, 2021-04-25:
Was this ever concluded? I have the same issue except when I run the console in Chrome I get this outut:

jquery.min.js?v=0.42.1:5 POST http://192.168.1.151:8765/config/0/set/?_=1619370786448&_username=admin&_signature=f5e00b63e26e5f9293d5c6b4600baeed2337b23b 500 (Internal Server Error)
send @ jquery.min.js?v=0.42.1:5
ajax @ jquery.min.js?v=0.42.1:5
ajax @ main.js?v=0.42.1:477
actualApply @ main.js?v=0.42.1:2627
doApply @ main.js?v=0.42.1:2695
(anonymous) @ main.js?v=0.42.1:926
dispatch @ jquery.min.js?v=0.42.1:4
v.handle @ jquery.min.js?v=0.42.1:4

john72oliver, 2021-04-25:
And this is in my daemon.log:

Apr 25 18:20:52 raspberrypi meyectl[359]:    ERROR: motion not running, starting it
Apr 25 18:20:53 raspberrypi meyectl[359]:    ERROR: failed to start motion: motion failed to start
Apr 25 18:20:53 raspberrypi meyectl[359]: Traceback (most recent call last):
Apr 25 18:20:53 raspberrypi meyectl[359]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 325, in checker
Apr 25 18:20:53 raspberrypi meyectl[359]:     motionctl.start()
Apr 25 18:20:53 raspberrypi meyectl[359]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 129, in start
Apr 25 18:20:53 raspberrypi meyectl[359]:     raise Exception('motion failed to start')
Apr 25 18:20:53 raspberrypi meyectl[359]: Exception: motion failed to start

So when I ran this:
sudo meyectl startserver -c /etc/motioneye/motioneye.conf

I got this:

INFO: hello! this is motionEye server 0.42.1
Usage: motion [OPTIONS] COMMAND [ARGS]...
Try 'motion --help' for help.

Error: no such option: -h
   ERROR: motion failed to start
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 334, in start_motion
    motionctl.start()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 129, in start
    raise Exception('motion failed to start')
Exception: motion failed to start
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 439, in main
    run()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 395, in run
    application.listen(settings.PORT, settings.LISTEN)
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2042, in listen
    server.listen(port, address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 143, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 168, in bind_sockets
    sock.bind(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

silversword411, 2021-04-26:
I closed this issue since it seemed to be device/browser specific and I couldn't replicate reliably. Have you tested on two different devices and found consistent issues?

john72oliver, 2021-04-26:
I've tried it with two different pi's and it's exactly the same. Looking at the error above it seems to be trying to listen on a socket that's in use. This seems to be preventng Motion from running:

 systemctl status motioneye.service
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-04-25 18:30:56 BST; 22h ago
 Main PID: 1392 (meyectl)
    Tasks: 5 (limit: 3860)
   CGroup: /system.slice/motioneye.service
           ├─1392 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─1403 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Apr 26 17:10:38 raspberrypi meyectl[1392]:     raise Exception('motion failed to start')
Apr 26 17:10:38 raspberrypi meyectl[1392]: Exception: motion failed to start
Apr 26 17:10:48 raspberrypi meyectl[1392]:    ERROR: motion not running, starting it
Apr 26 17:10:48 raspberrypi meyectl[1392]:    ERROR: failed to start motion: motion failed to start
Apr 26 17:10:48 raspberrypi meyectl[1392]: Traceback (most recent call last):
Apr 26 17:10:48 raspberrypi meyectl[1392]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 325, in checker
Apr 26 17:10:48 raspberrypi meyectl[1392]:     motionctl.start()
Apr 26 17:10:48 raspberrypi meyectl[1392]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 129, in start
Apr 26 17:10:48 raspberrypi meyectl[1392]:     raise Exception('motion failed to start')
Apr 26 17:10:48 raspberrypi meyectl[1392]: Exception: motion failed to start

This is a bit over my head though.

john72oliver, 2021-04-26:
Apologies, this was my doing. 

I'm very new to Linux and when I read the begining of the install instructions it says to go to another page for general installation. That page says that I need a few things installed so I installed them. 

Then I went back to the main installation instructions and blindly followed them. What I did there was installed a few things twice which would explain why Motion is trying to listen on a port which is already in use. Lesson learned, read everything carefully. 

Thanks anyway.