motioneye-project/motioneyeos #670: Can't login over wifi

sktrgk, 2016-12-29:
MotionEyeOS Version: 20161027
Board: Rasberry Pi 3
Camera: Raspberry Pi Camera V1

I am able to login using both user and admin credentials (using default user/pass and with a set password) when connecting to MotionEye over wired connection. I can connect to the Pi using the built in Wifi no problem, but when I try the same user or admin credentials to login, it always says "Invalid Credentials", no matter what I try. 

Is this a built in security feature? Or are there some settings I'm missing? I've attached the motion log where it describes the failed attempts. Thank you.

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/677592/motioneye.txt)

ccrisan, 2016-12-30:
motionEye web app doesn't distinguish between wifi or wired connection so there shouldn't be any difference between the two connection types. Make sure you don't accidentally access the IP address of another motionEye(OS) unit.

sktrgk, 2016-12-30:
Thanks for the response. I didn't think it would distinguish between the two. I only have this single motionEye(OS) unit on my network. And it still fails to login when I go to the IP for the wifi adapter. Whether or not the wired network is plugged in. 

What else can I check to try and debug the issue?

ccrisan, 2016-12-31:
Have you tried clearing the browser cache? Cache data may be associated with one IP address but not with other.

sktrgk, 2017-01-03:
Yup, that was the issue. I feel silly for missing that. Thanks for walking me through to the solution.

closed 2017-01-03 by sktrgk