motioneye-project/motioneyeos #671: motioneyeos variants

xblox, 2016-12-30:
Hi,
not an an issue but would you be interested in forks and promoting them also ? Right now I am building some other things into motioneyeos: MQTT (really important to wire it to all that IoT stuff without hassle) and my own components: "xblox" (visual block language to build stuff with GPIO and anything else commandable). 

My  main reason behind making a MotionEye-OS fork is because it would be a real pity to waste a whole 
30 $ board for a single functionality when it could still run some home-automation or connect other sensors and what not. (we have a lot of use here for a NAS/Video-Surveillance/IoT combo)

And, are there plans for plugin system? Right now I am trying to integrate your web api in my "xblox" language but somehow I'd prefer to build my stuff directly into your system instead of the other way around. 

Anyhow, how about a some MotionEyeOs builder service ? For instance people can choose to have nginx,php, nodejs,.... ? That would be awesome. Just don't know how easy it would be to have a web-front end for 'buildroot". At the end it would be so much easier to have MotionEyeOS as deb file just...so people can plug it just.

Thanks for any input, just bumping here. 

guenter

[image]

[image]

closed 2016-12-31 by xblox

ccrisan, 2016-12-31:
motionEyeOS is basically an underlying OS + motionEye. It's impossible (or at least it does not make any sense) to have an operating system provided as a .deb file.

If people want to add custom packages to the OS, the recommended way is to start with their favourite Linux distro and install motionEye on top of it, along with any extra stuff they need.

Advanced users could configure motionEyeOS (like you said) by adding or removing packages as they please, using BuildRoot's configuration tool. The bad news is that it must be compiled after that.

xblox, 2016-12-31:
ok, thanks.

xblox, 2016-12-31:
just in case you are interested, I am maintaining the fork [here](https://github.com/xblox/motioneyeos#a-motioneyeos-fork).

ccrisan, 2016-12-31:
This is actually a nice idea. What we could do to merge our efforts is maintain two separate configs for each board: one for the small (actual version) and one with many packages, where users should accept the idea of having a relatively large OS image with longer upgrade duration and potentially slower startup time.

xblox, 2016-12-31:
sounds good to me. I am still in the learning phase for BuildRoot (awesome) and I want to find a few new built-in candidates to make better use of the upcoming paho MQTT client, eg: "node-red" (so users have lots of more notification possibilities,...). I will try get this as board - config/overlay as you said and let you know. 

I do of course understand your concerns regarding size and overhead. 

Finger cross and thanks !