motioneye-project/motioneyeos #672: MQTT

xblox, 2016-12-30:
Hi,
would you accept a PR adding the python paho mqtt client, and of course some ui for the web - app ? 
g

would be great to know about some code style/contribution guide or editor configs btw.

ccrisan, 2016-12-31:
I believe this is related to https://github.com/ccrisan/motioneyeos/issues/671. I am usually happy to receive, review and merge PRs. But when it comes to adding stuff to the motionEyeOS image which I do my best to keep small, I tend to filter packages.

motionEyeOS should be seen as a firmware for camera devices. I don't expect users to code too much against it and configuration should be as simple as possible.

This being said, I don't think an MQTT client is something that would interest many motionEyeOS users. If you want to maintain an extended fork of motionEyeOS, with many other packages that users request, I'd be happy to help. Otherwise, you could always go with your favourite Linux + motionEye.

closed 2016-12-31 by ccrisan

xblox, 2016-12-31:
I can't really agree with your reasoning but ok, at least I tried to make this here better.

ccrisan, 2016-12-31:
I can understand that you don't agree with me, but I'm expecting a reason :) What exactly do you not agree with?

xblox, 2016-12-31:
well, the Paho mqtt client makes a few kb more only and it would open at least the doors to connect MotionEyeOS to some other systems than only relying on "cmd" for Motion triggered. But ok, don't bother. I do this in a fork now :-)
g

xblox, 2016-12-31:
btw, MQTT would even trigger if the host went down (called "last will") and so a consumer knows something is very wrong at his place. I am doing this here because lots of people (including my taller) got rubbed in our village and I am working out a reliable solution, especially for the case the camera system went entirely off (the thieves indeed cut power and phone cables first).

ccrisan, 2016-12-31:
The biggest problem I think is the lack of a motionEye API. You won't be able to properly use MQTT with motionEye without setting a wrapper script for all events.

xblox, 2016-12-31:
yeah, that's something I am investigating this days. Not sure where to begin, I admit. At the end I want 2 MQTT messages:

/MotionEyeOS/motionDetected with 

payload:{
  camera:'name',
  time:'...',
  image:'base 64 image' // would be cool,
  host:'some host id'
}

/MotionEyeOS/motionStopped with
payload:{
  camera:'name',
  time:'...',
  host:'some host id'
}

Aren't you calling some user shell command when it's triggered ? That would be the perfect place to activate the MQTT client.  

anyhow, I will find it! Thanks already for your heads up on this, despite we should relax on the sofa this days.

ccrisan, 2016-12-31:
Well as I said, you could manually configure wrapper scripts for the *Run A Command* options. There's no other way I can think of.

I have proposed an [API](https://github.com/ccrisan/motioneye/wiki/API-(Draft)) for motionEye but I haven't had the time to implement it and to be honest, I have other priorities these days, therefore I can't promise anything. But, in the long term, this should be the way an MQTT "agent" should interact with motionEye.

xblox, 2017-01-01:
hi,
yeah, the "Run A Command" would be some place but not really my cup of tea. I prefer to properly extend ui settings and forward from [RelayEventHandler]( https://github.com/xblox/motioneye/blob/master/motioneye/handlers.py#L1668) to MQTT (client should be started together with meyectl) . Unfortunately my Python is pretty rusty and I'd have preferred some plugin system but ok. Fine for me :-)

thanks again.

xblox, 2017-01-01:
ok, [basics done](https://github.com/xblox/motioneye/blob/master/motioneye/handlers.py#L1836). It works fine so far:

[image]

<hr/>

[image]

the mqtt client works fine and stable, adds just a few kb to the image.

xblox, 2017-01-01:
yeah, i would think so. Let me test it in the field next week and polish it a bit. I will push a PR right away then (only impacts motionEye, not motionEyeOS).

btw, I was adding the MQTT settings in the ui and noticed quite a few code styling issues but also structural things. I actually would love to add "require-js" and "lodash" and split all separated in modules. I know you probably want to keep it simple but its a sure thing that I am going to extend motionEye(server/client) project quite a lot over the next weeks (localization, and some ACL). If you like, I can make another PR after you processed the other open PRs. I also don't mind to re-write/port your existing JS over to Typescript (doing this to most of my projects/libs right now), let me know. 

thanks for the heads up and no worry, I see you're busy man :-)

ir-fuel, 2019-10-12:
Did something actually happen with this?

derekmerck, 2020-01-31:
## Setting up mqtt motion alerts on a Raspberry Pi MotionEyeOS Host

Partial "run a command" solution.  No last will and testament, etc. as requested by OP, but wanted to share for reference.

MotionEyeOS has no package manager and a read-only file system, so this is a side-load hack and ymmv.

1. Download the latest Debian armhf packages for [libmosquitto1](https://packages.debian.org/sid/armhf/libmosquitto1/download) and [mosquitto-clients](https://packages.debian.org/sid/armhf/mosquitto-clients/download)
2. Unpack each package with `ar x file_xxx.deb` (requires Mac or Linux machine)
3. Unpack the resulting `data.tar.xz` file
4. Move the lib and bin files to the meye host's `/tmp` directory with something like `scp usr/bin/mosquitto_pub usr/lib/libmosquitto.so.1 admin@<MEYE_HOST>:/tmp`
5. The "run a command" invocation to publish is `LD_LIBRARY_PATH=/tmp /tmp/mosquitto_pub -h <BROKER_HOST> -t cameras/<MY_CAMERA>/motion -m ON`
6. You can login remotely with `ssh` and test it

I was somewhat surprised that this worked!

Thanks to @ccrisan for sharing your work on this project!

jshep321, 2020-03-25:
I tried the above but was missing libssl also.  Did you have to install other libs?  Note I was on a very old motioneyeOS version and my upgrade failed...  So this experiment is halted for now. :)

derekmerck, 2020-03-25:
Doesn't surprise me with an older version.  I am using `motionEyeOS 20190911`, fwiw.  I am sure that I didn't have to mess with `libssl` b/c I probably would have just given up!  

Is it worth trying with the latest motionEyeOS and reconfiguring your cameras?  If you are doing something embedded, it may be as easy as just swapping out the SDHC card and swapping back if it doesn't work.

jshep321, 2020-03-29:
Hmmm odd... Upgraded to the new version 20190911 and now it gives me this: 
`/tmp/mosquitto_pub: /lib/libc.so.6: version GLIBC_2.28 not found (required by /tmp/libmosquitto.so.1).  `