motioneye-project/motioneyeos #674: web hook clarifications

pantchox, 2017-01-01:
Hi,
very nice project! currently i am testing some of its features.

the "Call A Web Hook" feature looks cool but there is not documentation on the "HTTP Method" usage.
there are 4 options
- GET -> which is obvious 
- POST (query) -> is it like get but post eventually (instead of url query params its body request params)
- POST (form) -> what does it even mean?
- POST (json) -> does it post the query params as JSON object, if yes which post params the JSON object is belonging to?

Would like some clarifiaction, also knowing if there is ability to add/remove fields and also attach the images that were recorded in the motion triggered, same as the email notifications.

thanks!

ccrisan, 2017-01-01:
I admit this functionality requires a bit more details.

1. Setting it to `GET` and the URL to `http://www.example.com/path/?arg1=value1` has the obvious effect.

2. Setting it to `POST (query)` and the URL to `http://www.example.com/path/?arg1=value1` has the effect of using POST with an empty body, while keeping the arguments as query arguments (as if it were a GET).

3. Setting it to `POST (form)` and the URL to `http://www.example.com/path/?arg1=value1&arg2=value2` has the effect of using POST with a form url-encoded body (application/x-www-form-urlencoded), taking name-value pairs from the given URL (e.g. `arg1=value1&arg2=value2`).

4. Setting it to `POST (json)` and the URL to `http://www.example.com/path/?arg1=value1&arg2=value2` has the effect of using POST with a JSON body (application/json), taking name-value pairs from the given URL and sending them as a dictionary (e.g. `{"arg1": "value1", "arg2": "value2"}`).

I know it's not obvious and not very flexible and far from ideal, but this is what I was able to come up with without spending too much time on the feature.

If none of the 4 cases is good enough, the `curl` way wrapped into  a script will always work.

I hope this helps.

pantchox, 2017-01-01:
thank you for you kind response! this should be added to the documentation.

closed 2017-06-01 by bortek

swissglider, 2017-10-28:
Hi and how to add the special tokens ?

thathollywood, 2018-03-10:
Thanks. Just what I was looking for 👍  Sending webhook to Slack