motioneye-project/motioneyeos #677: Editable Mask not working version 20161212

martinrgmiller, 2017-01-02:
motionEye Version 	20161212
Motion Version 	4.0.1+git37b3595
Platform: Raspberry Pi 3
Image Resolution : 1920x1080
90deg rotation

Mask as designed and applied: 
         http://prntscr.com/dqkzvh
When reloaded this is how the mask is displayed. 
         http://prntscr.com/dql040

It looks like the recording of the editable mask is at a lower resolution than the original or perhaps rotated and squashed. 
The mask is designed to remove false positives from clouds and tree moving in the wind. However, the weather is unseasonably good so I can't actually tell if the mask is working or not.

Chiny91, 2017-01-02:
I've not used an edit'able mask since motionEyeOS introduced it (!), so not sure how you have made your mask.   If made with separate software, I can say that _motion_ is very strict on what it accepts and it is worth trying a different package.

Also, have a look at #89 where I note how to check whether or not a mask is working.

Have you tried smart mask for the trees ?   Not sure it would deal with clouds though.

martinrgmiller, 2017-01-02:
I've used the internal mask editor in the newer version of motioneyeos. This is accessed via the 'edit mask' button in the setting. The issue appears to be the OS not recording the mask file correctly. 
In previous versions I've used the mask feature via the file method, if you pass the maskfile parameter to the newer version it seems to convert it to the new system.

ccrisan, 2017-01-02:
@martinrgmiller this is definitely not how it is supposed to work. It may be because of the 90 deg rotation. I'll try to reproduce the issue. In the meantime, can you reproduce this behaviour? Does this happen all the time when you play with the mask?

martinrgmiller, 2017-01-02:
@ccrisan thanks for the prompt reply. I can confirm that the issues is repeatable.
I tried chaning the rotation to 270 (from) and the problem reproduces the same. see
http://prntscr.com/dqqwoe
http://prntscr.com/dqqwzu
There isn't any change in the scaling of the mask.

ccrisan, 2017-01-24:
@martinrgmiller what if you redesign your mask *after* you have rotated your camera. Does the issue persist?

martinrgmiller, 2017-01-24:
@ccrisan 
ok i set the image rotation to 0deg and 
set: [](http://prntscr.com/dzvs4z)
after applied: [](http://prntscr.com/dzvsmb)
the one thing I have noticed here is that the 0deg rotation mask has only 18 x 32 squares (hxw)
after image rotation to 90deg the mask looks like
[](http://prntscr.com/dzvtz0)
the number of square in the mask has higher and the mask isn't rotated.

also the **message.log** file has this entry 

```
20:59:54 Cam1 user.warn motion: [1:ml1] [WRN] [ALL] get_pgm: The mask file specified is not the same size as image from camera.
Jan 24 20:59:54 Cam1 user.warn motion: [1:ml1] [WRN] [ALL] get_pgm: Attempting to resize mask image from 1920x1080 to 1080x1920
```
my camera image is 1920x1080 rotated so this should be ok. 

with the rotation set to 90deg and if i change the mask design to
[](http://prntscr.com/dzvw72)
and save and apply the mask, i get 
[](http://prntscr.com/dzvwqh)
this is the same shrinking behaviour and the log reports the same error.
```
Jan 24 21:05:30 TortoiseCam1 user.warn motion: [1:ml1] [WRN] [ALL] get_pgm: The mask file specified is not the same size as image from camera.
Jan 24 21:05:30 TortoiseCam1 user.warn motion: [1:ml1] [WRN] [ALL] get_pgm: Attempting to resize mask image from 1920x1080 to 1080x1920
```

ccrisan, 2017-01-25:
Please do *not* rotate the image *after* editing the mask.

martinrgmiller, 2017-01-26:
@ccrisan my original tests were done with the image rotation before designing / applying the mask. My last post was to show that the same shrinking error occurs regardless of when the rotation is applied.
Just to be absolutely clear, here's another example when I start with the image in the required rotation (90deg).
here's the mask after I've edited it and saved (but not applied)
[http://prntscr.com/e0t9q8](http://prntscr.com/e0t9q8)
after i press apply, and go back to check / edit the mask it looks like this.
[http://prntscr.com/e0taaj](http://prntscr.com/e0taaj)
Basically, when image is being reduced into the wrong dimensions.

When I start with the image not rotated, the problem does not occur but the mask resolution is more coarse as stated in my previous post (18x32 squares over the 1080x1900 image).
[http://prntscr.com/e0tc59](http://prntscr.com/e0tc59)
My only work around here is to mechanically rotate the camera and tolerate the low fidelity mask.
Obviously this doesn't make watching the camera or reviewing the images easy but the mask doesn't change in size.

I'd be interested to know it you can see the same effect. Especially in the mask fidelity and shrinking.

closed 2017-06-01 by bortek