motioneye-project/motioneyeos #678: No stream for default user.  version 20161212

martinrgmiller, 2017-01-02:
motionEye Version 	20161212
Motion Version 	4.0.1+git37b3595
Platform: Raspberry Pi 3

I don't have a password set for the default 'user' so I'm expecting the motioneye camera webpage to display the camera webpage when first loaded.
Instead it shows the spinning 'loading' logo. 
      http://prntscr.com/dql8tc

If I force the default user to login by clicking on the 'switch user' icon atthe top and entering 'user' and no password there is no change and the logo keeps 'loading'.
If I force the user to have a password, it still doesn't show the camera feed.

The only way to see the camera is to login as 'admin'.

The work around for a normal user to see camera feed is to enable 'streaming' and directly access the feed via the unique port. However this does mean i can't have an easy portal to monitor several camera (my plan eventually).

ccrisan, 2017-01-02:
Make sure you haven't set the "Frame Rate Dimmer" to 0 in your user's preferences.

martinrgmiller, 2017-01-02:
I wasn't aware of changing those setting. Thanks for the quick fix.

closed 2017-01-02 by martinrgmiller