motioneye-project/motioneyeos #679: Native support for RTSP

cl0udb0y, 2017-01-03:
Hi ... do you have any plans to support any other protocols other than HTTP for camera streams? RTSP would be a personal preference...

ccrisan, 2017-01-03:
RTSP is already supported for more than one year now.

closed 2017-01-03 by ccrisan

cl0udb0y, 2017-01-03:
is there a doc/readme you can point me to as using rstp://.... in the URL just gives 'RTSP' not supported... is it only supported on the MotioneEyeOS as I have the MotionEye on a separate ubuntu server and just using RPi as the cameras along with Panasonic and Hikvision for others?

ccrisan, 2017-01-03:
Well, then you're on the wrong project. This is motionEyeOS. For motionEye, you must make sure you have [motion 4.0](https://github.com/Motion-Project/motion/releases) installed.

cl0udb0y, 2017-01-03:
Thanks.. I'm using both, thanks for the pointer Calin. Rgds John