motioneye-project/motioneyeos #680: FYI, not really in issue. Configure motioneye for multiple wifi

RudyRipDraw, 2017-01-04:
This is not an issue but rather FYI to help others.

System is a Raspberry Pi zero/ Raspberry Pi B.

I used Raspberry Pi B for to develop and troubleshoot the system before transferring the SD card to the Pi zero

I have two wifi systems in the house and previously was only able to get motion eye to work on the ethernet and wifi A but not wifi B.  Wifi B is generated thru a Linksys Powerline Network Extender.

When in the wifi B area with the ethernet connected, I would use the motioneye gui and enter in the wifi B  network name and network key.    However wifi B only has a passphrase, not a network key.  Wifi A (older router) has an actual network key.  I am not sure if this is part of the problem.

Instead, I modified the /data/etc/wpa_supplicant.conf file directly and now my Raspberry Pi zero based MotionEye works on either wifi without any modifications.  You just need to reboot when the system changes between physical area between the two wifi.

[image]

NOTE, the image is "clipped" and the full ssid and psk are NOT shown.

The first network is an actual network key, while the second network uses a passphrase.  Could it be that the MotionEye gui cannot accept/translate a passphrase?  

Again, this is NOT an issue, but rather a FYI in case other run into this problem.

Thanks for your work on Motion Eye!!!

worldburger, 2017-05-23:
This may actually be considered an issue for some users.

@ccrisan is there a way to have mobileeyeos automatically scan and reconnect to available wifi from the wpa_supplicant.conf file if the wifi connected at boot is no longer available?

ccrisan, 2017-05-23:
You can set up as many networks in your wpa_supplicant.conf (and you'll get the roaming functionality you want) but don't expect motionEye's UI to allow you to manage it anymore. You'll probably want to comment out [wpa_supplicant_conf](https://github.com/ccrisan/motioneye/wiki/Configuration-File#wpa_supplicant_conf) in your [motioneye.conf](https://github.com/ccrisan/motioneyeos/wiki/motioneye.conf) file.

closed 2017-06-01 by bortek

vaggelis305, 2021-01-20:
Opening