motioneye-project/motioneyeos #681: Unable to use two 1080p cams with one RPI3 ?

remizik, 2017-01-04:
Hello,

I have a RPI3 with the last version of motioneyeOS installed.
I also have two of these cameras : https://www.amazon.fr/gp/product/B01JG43TD0
I'm using a 4Gb microSD card (it might be slow !).
Also using a 2A power supply.

Apparently the RPI3 cannot handle two cams at 1080p :( One works ok.
When I go down to 1280*1024px resolution on both cameras, it works (at a very sad framerate though).

both cams at 1920*1080px : http://i.imgur.com/b2d1oMy.png
both cams at 1280*1024px : http://i.imgur.com/EkDcRiH.png

My question is, is there anything I can do to improve/tune the settings to be able to use the full resolution of these cameras ?

I was planning on using 4 of these cameras on one RPI3 but I guess this won't happen ... :(

Thanks !

specs of the cameras: 
_2.0Megapixel résolution 1080p camera usb cmos 3.6mm lens 
Capteur: 1 / 2,7 "CMOS OV2710
1920 (H) x 1080 (V) pixels / Max. Résolution 1920X1080
Format d'image: MJPEG ou YUY2 option
USB Protocal: USB2.0 HS / FS
Exposition automatique AEC: Support
Auto blanc Blance AEB: Support

Pixels efficaces:
1920 (H) x 1080 (V) pixels/1280 (H) x 1024 (V) pixels/
1280 (H) x 720 (V) pixels/1024 (H) x 768 (V) pixels/
800 (H) x 600 (V) pixels/640 (H) x 480 (V) pixels/
352(H) x 288 (V) pixels/320 (H) x 240 (V) pixels
Performance:
1920 (H) x 1080 (V) pixels MJPEG 30fps YUY2 6fps ;1280 (H) x 1024 (V) pixels MJPEG 30fps YUY2 6fps
1280 (H) x 720 (V) pixels MJPEG 60fps YUY2 9fps ;1024 (H) x 768 (V) pixels MJPEG 30fps YUY2 9fps
800 (H) x 600 (V) pixels MJPEG 60fps YUY2 21fps ;640 (H) x 480 (V) pixels MJPEG 120fps YUY2 30fps

Paramètres ajustables: Luminosité / Contraste / Saturation des couleurs / Définition / Gamma / WB
tension: DC 5V vcourant: 150mA
température de travail: DEGRES (-20 ~ 70)
Soutien OS: "WinXP / Vista / Win7 / WIN8 / Linux avec UVC / Mac OS X 10.4.8 ou ultérieur / WinCE avec UVC / Android 4.0 ou au-dessus"_

remizik, 2017-01-09:
:(

is anyone able to use two 1080p camera at full resolution on one RPI3 ?

djjayrock, 2017-01-18:
I have these cameras.  I have the same issue.  I also have a problem with jumping video

george-vieira, 2017-02-08:
tried overclocking? it may be purely the pi's aren't cut out for the job cpu bound wise. have you tried running top and seeing the 2 cameras are using separate cores and that it's not a ethernet bottleneck for whatever reason?
I have mine dumping their recording to an NFS4 mount which works well so i dont have SD card as a limitation.

closed 2017-06-01 by bortek