motioneye-project/motioneyeos #683: Feature request: Block IP after failed logon-attempts?

Goudtik, 2017-01-06:
Is there a possibility to block a incoming IP-address after a few failed login attempts on MotioneyeOS?
Maybe by using fail2ban or something? It would really improve security!

skinnyjay, 2017-01-06:
Just tested and done, found using google already written out in an article and packaged into Raspbian. Works just fine: https://dvpizone.wordpress.com/2014/03/02/how-to-install-fail2ban-on-a-raspberry-pi/

Goudtik, 2017-01-06:
but will this work under MotioneyeOS? (currently i'm using Motioneye under raspbian but i'm considering to abandon this and install the OS version)

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

rotrials, 2019-03-05:
@jarednthomas but the configuration from the link is for ssh not for motioneye login...