motioneye-project/motioneyeos #688: Odroid C2 not booting with released images

ramaxcalmax, 2017-01-08:
Hello Calin,

first of all thanks for your work on motioneye!!!

I've got it running on my Raspberry Pi 2 and wanted 
to get it running on a more powerfull Odroid C2.

But with the released images my odroid just don't boot up.
Red LED is on and blue one is on for about 3 seconds and the 
starts blinking:  "blink-blink...1 sec...blink-blink..." aso.
after about 30 secs it starts again with the 3 sec lighting of the blue LED.
The network LEDs shows activity until it "reboots?" and the starting again.

There is no meye-____ showing on my router (except the already running Raspberry Pi 2).
I've tried booting it up with only power and network connected and then with 
additional HDMI->HDMI Monitor, Keyboard, Mouse and USB-Memory-Stick.

Android 5.1 and Ubuntu 16.04 are running fine on the Odroid (with the same 
burning-method and sd-card).
Other images (Open Media Vault/Openelec-Image) are not running at all, too.

Do you have a hint what's going wrong?

Many thanks in advance for your help.

kind regards
ramaxcalmax

ccrisan, 2017-01-08:
Here are a few troubleshooting ideas:
* attach a USB to serial 3V3 cable to Odroid's serial pins and see what it has to say at boot time (don't bother connecting a monitor, as Odroid guys were kind enough not to provide a framebuffer driver)
* try older OS images and see if any of them works

ramaxcalmax, 2017-01-09:
Hi Calin,

thanks for the help!
I already tried the 4 last released versions with different
SD-Cards and even different writers and different burning software...

Sorry but I'm not that deep in the matter, so reading out the pins while
the thing is booting, is behind my horizon.

Any other ideas or do you know someone with a running C2
(so I can "copy" his/her setup)?

Thanks!

ramaxcalmax, 2017-01-09:
Another thing:

I've installed the Motioneye to the ubuntu system.

After the install and restarting it I found the meye on the router
but after a restart of ubuntu I wasn't even able to get the motioneye
running again...

Isnt there an Icon in the startmenu?
What to do to get it automatically starting with ubuntu?

ccrisan, 2017-01-09:
On Ubuntu, see https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu. Step number 7 will help starting motionEye automatically at boot.

P.S. on Ubuntu, motionEye has nothing to do with your router or the IP address it receives.

closed 2017-06-01 by bortek