motioneye-project/motioneyeos #689: Request: Trigger alarm when motion detected

umichfan1, 2017-01-08:
motionEyeOS is an *amazing* project. Thank you very much for designing it, ccrisan. I am wondering if there are any plans to make it possible to trigger an alarm when motion is detected. I know that the action buttons allow an alarm to be triggered when a button is pressed, but it would be much better if the alarm were triggered automatically when an intruder is detected. Thank you.

ccrisan, 2017-01-08:
Use the Motion Notifications settings to configure your alarm.

closed 2017-01-08 by ccrisan

umichfan1, 2017-01-16:
Thanks for the reply. Would you mind pointing me in the right direction to
figure out how to play an audio file through the audio jack using the
MotionEyeOS operating system? Thank you.

Chris

tomwoodhams, 2017-02-13:
Yes, I want to do this too, but can't work it out.

molly78, 2017-02-20:
Try the command 

ffplay /path/to/example.mp3

jdaien4783, 2018-05-28:
@umichfan1 or @tomwoodhams did you ever figure this out. I also want to do this.  Did the suggestion @molly78 used work?

umichfan1, 2018-05-28:
No, unfortunately I never did betty this to work. In the end i used two
raspberry pi's: one for capturing the video, and the other for triggering
the alarm.

jdaien4783, 2018-05-28:
Ok thanks anyway. I thought about using two raspberry pi’s as well. How did you do it? Which OS did you use on the send pi?

Thanks,

Josh

umichfan1, 2018-06-02:
Hi Josh,

If I remember correctly, I used this as a starting point:

http://www.whatimade.today/quick-effective-low-cost-home-protection-with-raspberry-pi-and-esp8266/

I just used Raspian on the second Pi. Looking back at my notes, I also
found these links helpful:

https://www.raspberrypi.org/learning/parent-detector/worksheet/

https://www.raspberrypi.org/forums/viewtopic.php?t=30623&p=266493

Good luck!

Chris