motioneye-project/motioneyeos #690: Added 2nd camera, Shows Grey then Recovers repeatedly

skinnyjay, 2017-01-08:
I've had no issues running Camera1, an HP Desktop Webcam, however I recently added my second camera, a Playstation Eye Camera, and it works for a few minutes before going grey, and showing:

Unable to Open Video Device

After a few moments, video returns, but it's a reoccurring issue for me. This is on a RPi3

cap.driver: "ov534"
cap.card: "USB Camera-B4.09.24.1"
cap.bus_info: "usb-3f980000.usb-1.5"
cap.capabilities=0x85200001
------------------------
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1:ml1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] v4l2_select_input: name = "ov534", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Config palette index 8 (MJPG) doesn't work.
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (0) YUYV (YUYV 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 0 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1] [NTC] [VID] v4l2_set_pix_format Selected palette YUYV
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YUYV (640x480)
[1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
VIDIOC_S_FMT: : Device or resource busy
[1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
[1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1:ml1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1:ml1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "ov534"
cap.card: "USB Camera-B4.09.24.1"
cap.bus_info: "usb-3f980000.usb-1.5"
cap.capabilities=0x85200001
------------------------
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1:ml1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] v4l2_select_input: name = "ov534", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Config palette index 8 (MJPG) doesn't work.
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (0) YUYV (YUYV 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 0 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1] [NTC] [VID] v4l2_set_pix_format Selected palette YUYV
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YUYV (640x480)
[1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
VIDIOC_S_FMT: : Device or resource busy
[1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
[1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1:ml1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1:ml1] [NTC] [ALL] motion_loop: Video signal lost - Adding grey image
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1:ml1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "ov534"
cap.card: "USB Camera-B4.09.24.1"
cap.bus_info: "usb-3f980000.usb-1.5"
cap.capabilities=0x85200001
------------------------
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1:ml1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] v4l2_select_input: name = "ov534", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Config palette index 8 (MJPG) doesn't work.
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (0) YUYV (YUYV 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 0 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1] [NTC] [VID] v4l2_set_pix_format Selected palette YUYV
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YUYV (640x480)
[1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
VIDIOC_S_FMT: : Device or resource busy
[1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
[1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1:ml1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [NTC] [VID] vid_close: Closing video device /dev/video0

skinnyjay, 2017-01-08:
Additionally, I only notice this issue when trying to run the Playstation Eye at 640x480.
I do not have these issues at 320x240, but the quality is so poor I might as well not have the camera up at all.

oysterbomb, 2017-01-08:
I've been having the same issues with other makes/models of cameras.  So far i've narrowed it down to the cameras requesting to use more bandwidth on the USB hub than is available.

skinnyjay, 2017-01-08:
Setting palette to YUYV seemed to fix the issue for now (15)

closed 2017-06-01 by bortek