motioneye-project/motioneyeos #695: Hard to set up the motion trigger.

JacquesBB, 2017-01-12:
Hi, 

Thanks for this software which is close to what I was looking for. 
I dont know whether this is the proper place for asking, but I have serval issues of various importance. 

- I cannot manage to make the motion trigger properly work. 

My settings 
- frame threshold : 1%
- auto noise : No 
- light switch : no
Motion Gap : 10s 
Captured Before	:10
Captured After	:10
Minimum Motion Frames :3

I have also a mask .

First question : Is the frame threshold  a percentage of the full picture, or a percent of the unmasked  (active) section of the picture ? 

B) WHen I finally got something triggered, several times I got  just an email, with no pictures. 
I have cleaned all my files.  Today I received a single picture, but green with black zone 
solarised with the detection. Is it possible to get also a normal picture . 

C) Unrelated : How can I get a snapshot of the frame when monitoring live. If not possible, take it as a feature request, to have a button to press for capturing either still picture or recording video. 

D) I tried a full install of motioneye over raspbian, with only motioneye on a fresh 
raspbian install. It was working, but was unstable. I went back to motioneyeos. Is this frequent? 

Thanks for all

n2mfc, 2017-01-26:
I feel your pain.  I think I can give you one possible answer to your green with black zone picture, you may have the motion capture diagnostics turned on and that shows the mask and pixels that detected motion.

My experience has been that it is all trial and error.  I tried finding a manual that explained each setting in detail, but like your question about threshold, is it total pixels or only unmasked area.  I would also like to know what the numbers in the overlay mean.  Is the top right number the number of pixels that motion is being detected in?  Would be more useful if that number displayed the percentage that matches the setting slider.

If anyone knows where the settings for motion detection are documented in detail, please provide in a response.  I was thinking of trying to look in the source code but that may prove difficult.

Chiny91, 2017-01-26:
For the questions about _threshold_, how about the [v3 manual page](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionThreshold) which looks like the answer.

_motion_ has moved to v4 (MrDave) and there will be differences but v4 is less well documented, although [GitHub](https://github.com/Motion-Project/motion) can be useful if indigestible.   Still, the [motion guide](http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionGuide) is probably a decent starting place.

n2mfc, 2017-01-26:
Thanks Chiny91!  The last link looks the most promising.  I found [this](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#Options_Motion_Detection) which answered some of my questions but it does not appear to cover some custom things that are in motionEye.

JacquesBB, 2017-01-26:
Indeed,  what is not clear is the relation between the percent threshold of motioneye and the pixel threshold of motion.

ccrisan, 2017-01-26:
I believe [this](https://github.com/ccrisan/motioneye/blob/master/motioneye/config.py#L786) is what you're looking for.

JacquesBB, 2017-01-26:
Thanks 
the line you mention is 

        width = int(ui['resolution'].split('x')[0])
        height = int(ui['resolution'].split('x')[1])
        data['width'] = width
        data['height'] = height

        threshold = int(float(ui['frame_change_threshold']) * width * height / 100)

But this is indeed what seems to cause problem. If I understand well;  the percentage is then referred to the whole picture. So if one  select a zone, it decreases the frame_change_threshold that is the pixel count,  but the percent (threshold) is referred to the total surface of the view. 

Do I understand right, or I miss something ? 

Thanks

ccrisan, 2017-01-27:
@JacquesBB you are right, the formula doesn't take the mask into account. The code could be adjusted to take into account only the unmasked surface.

elstimate, 2017-02-14:
Hello,

I have issue related this topic.
using MotionEye vear: 20161212 and Motion ver. 4.0. git37b3595

the main problem is about a recorded video, triggered by motion detection.
the issue can be described as a kind of "lagging" issue. namely when I play the recorded file, it's like starting with a normal scene of the camera, then the moving object appears normally, moving normally for a few frames and then, skipping the record to a bit later. it's like a "jump" in the recording.... seems like a bit of lag appearing during recording and the moment when the ME system "buffering" or "doing issue" due to this buffering / lagging situation.

so, it's a bit unpleasent when I have the recording, got the movement detection, but I have no full footage of the situation, due to "jump" in the story. so therefore, a short "action" cannot be analysed or so.

here is the setup:
motion gap (20) 
captured before (60) 
captured after (20) 
minimum motion frame (10)
[i've tried so many different combination with these settings above, did not solve issue].
camera is a LAN camera with great performance. frame rate set to 20 fps, res. 640x480
no performance issue with LAN IPCAM separately (e.g. checking video stream with VLC network player.)

please, give some help. if logs needed, please name them!

Thanks,
joe

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

eckirchn, 2019-04-28:
Joe - did you resolve this?   Note:  those units are in frames, not seconds..