motioneye-project/motioneyeos #697: MotionEyeOS

cam-chief-wally, 2017-01-12:
Buna ziua Calin,
in primul rind vreau sa-ti multumesc pentru munca depusa la “MotionEyeIS!
(first of all thanks for your work on MotionEyeOS!)
Sint “incepator” in domeniul “Raspberry Pi si de citeva zile ma ocup cu MotionEyeOS si sint foarte incintat pina acuma. In present as avea 4 intrebari de “incepator”:
1.	In momentul actual am o o lățime de bandă de descărcare de 40 MB / s, acest lucru nu ar trebui sa fie o problema pentru un flux fluid pentru Videoul produs de Camera Video. Din pacate fluxul Video nu este in intregime fluid, ci „transmisia” se face cu mici intreruperi ( smucituri ) FPS am reglat la Maximum posibil (30 FPS). Care ar putea sa fie cauza ? Este posibil ca puterea  prosesorului (CPU) de la Raspberry 3 sa fie prea slaba si / sau ar trebui sa dispun mal multa memorie Video ? (sau adaptorul de alimentare prea slab?)
2.	La un prieten care are si el MotionEyeOS se vede la “Videostream” in partea de jos temperature Procesorului si “Framrate” actual. Unde pot gasi aceasta setare? Eu gasesc numai Numele Camerei si Ora actuala.
3.	Ar fi posibil să se pună în aplicare un temporizator? Sunt conștient de faptul că Rasberry  nu poate fi pornit din nou, dar poate s-ar putea cel puțin închide si deschide prin setarea unui  temporizator camera video ?
Genial ar fi si o adaptare la timpii de răsărit și de apus alle soarelui, în funcție de poziția GPS (reglabil manual). Posibil ?
4.	 De cind va fi posibila incarcarea automata pe internet pe un „FTP” a „Still-Images”
 
Iti multumesc pentru eforturile tale si mult succes in viitor.

Multe salutari
Valeriu I.
*************************************************************************
Hello Calin,
First of all thank you for your work on MotionEyeOS!

I am new to Raspberry Pi and have been trying for a few days with MotionEyeOS and am very enthusiastic. Currently I have four beginner questions:
1. I have an upload bandwidth of 40 MB / s which is why this should not be a bottleneck for a maximum liquid video stream should be on the Internet. Unfortunately, the Videostream is nevertheless not liquid but jerks strong. The adjustable FPS I have in MotionEyeOS to the maximum. What could be jerky? Could it be due to the CPU power of the Raspberry 3 and / or can I still provide more graphics memory?
2. In the stream of a friend who also uses MotionEyeOS, the temperature of the CPU is shown on the lower left of the screen (the video stream). Where can I find this overlay setting? I find Framerate, Kameraname etc ... but not this temperature indication.
3. Would it be possible to implement a timer? I am aware that the Raspberry thus can not be switched on again but perhaps you could at least downshift or set via time switch, between which times the camera should be active? (Recording of photos in the interval is senseless, for example, in the dark). It would be an adaptation to sunrise and sunset time depending on the GPS position (manually adjustable).
4. When will the autom. Upload still images to FTP?
Many thanks for your efforts,

ccrisan, 2017-01-13:
Hi,

0. Please stick to English so that everyone can benefit of our discussion here.
1. It is known that the real performance on a RPi is a bit below of what users expect. It's usually CPU bound. However, I'd need to know what resolution and framerate have you set for your camera and what actual FPS you get.
2. It's under Expert Settings and it's called Enable Monitoring or something like that. It's only available on RPi.
3. This sounds just too complicated. No idea. I don't have too much time to spend on this project so the answer is probably *no*.
4. Some time in the future. I'm busy with other projects these days, but this is something I've been meaning to implement for a long time. No ETA though.

cam-chief-wally, 2017-01-13:
Hello together,
hello Calin,

now more details for my case, as screenshot.(Resolution:1024X768 / Framerate:30 FPS / actual FPS 12-20 FPS) Thank you for your friendly answers!
Maybe there is a well known good combination of the settings for a high-qualitative framerate and resolution on a Raspberri 3. F.e. forward grafik-power to the CPU...?!

Thank you very much,
greetings

PS: Looking forward to the FTP-Upload possibility in the future (without stress!)

Screenshots:
https://picload.org/image/rogwoici/pics1.jpg
https://picload.org/image/rogwoicw/pics2.jpg
https://picload.org/image/rogwoiir/pics3.jpg
https://picload.org/image/rogwopcp/pics4.jpg

closed 2017-06-01 by bortek